ataradov/edbg #3: Unknown Target Device: SAMD21 Xplained Pro

sabinoj, 2015-07-24:
I have been testing this with an Atmel SAMD21 Xplained Pro board I have. When I plug the board into the debug USB port (this is what the Atmel FAEs specified was the correct way to interface to the device for programming and debugging) with the "./edbg -l" command I get, "Attached debuggers:
  ATML2130021800014950 - Atmel Corp. EDBG CMSIS-DAP", but if I attempt to read the contents of the chip with the "./edbg -r" command I get, "Error: unknown target device (DSU_DID = 0x10010300)". I should point out that this board is less than a day old an I have not loaded any code onto it yet. I just wanted to test if this would be able to communicate to the device.

ataradov, 2015-07-24:
This is probably a new revision of the device. Just edit target_cm0p.c to add your device there. I'm intentionally not listing all potential devices, only ones that were actually tested.

The whole device detection is actually incorrect and is to be fixed in the near future. For now, just add your device to the list.

sabinoj, 2015-07-24:
Thanks for the quick reply. I updated the target_cm0p.c like you said and rebuilt the application. When I attempt to read with, "./edbg -br" it states:

Debugger: ATMEL EDBG CMSIS-DAP ATML2130021800014950 02.01.0157 (S)
Target type: Cortex-M0+
Target: SAM D21J18A-U
Reading...................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................Error: output file name is not specified

And when I attempt to specify an output file with, "./edbg -br output" it returns, "Error: malformed command line, use -h for more information".

ataradov, 2015-07-24:
Use -f option to specify a file.

sabinoj, 2015-07-24:
I did not notice the -f flag, thank you.

closed 2015-07-24 by sabinoj