ataradov/edbg #8: atmega support

aethaniel, 2015-09-18:
Hi @ataradov, some Atmel AVR boards (XMINI) embed an EDBG, well an mEDBG but with similar commands.
Do you think it would be easy for you to add their support?

Thanks,
T.

ataradov, 2015-09-18:
mEDBG is a completely different beast. As far as I can see it is not even CMSIS-DAP. And even with a real EDBG, I'm not sure if AVR programming commands are documented anywhere.

 This project will never support such a thing.

aethaniel, 2015-09-18:
sorry, I understand from this: http://www.atmel.com/webdoc/protocoldocs/ch01s01.html it was only a reduced (non-DGIable) EDBG, thus CMSIS-DAP. I will check the ones I have here on SAMD10-XMINI and atmegaxxx-xminis.
Indeed AVR programming commands are wrapped and don't appear in this document: http://www.atmel.com/webdoc/protocoldocs/ch02s02s01.html
I heard about a partial support of medbg in avrdude forum. Have to check.

ataradov, 2015-09-18:
Yes, it appears that mEDBG is CMSIS-DAP indeed. But I'm not about to include some proprietary reverse-engineered protocol. That's why I love ARM - they don't hide information on how to access their stuff. Other vendors should learn from this.

In any case, major part of this tool is SWD protocol, which would be completely useless. And I'm sold on ARM stuff, it is time to move on.

But I might try mEDBG for Xminis with Cortex-M MCUs.

aethaniel, 2015-09-19:
Understood, no problem for me regarding the focus on ARM devices.
By the way, your last comment made me think about another point:
I didn't checked really deeply this code but do you mean your implementation is mainly CMSIS-DAP+SWD, thus allowing the use of other CMSIS-DAP based tools?
I have discovered some device specific programming algorithm in target_cm*.c although I thought this was done inside the tool itself, as seen in https://github.com/mbedmicro/CMSIS-DAP

ataradov, 2015-09-19:
I eventually want to try other tools, but I don't know any cheap boards that 100% have CMSIS-DAP debugger. ST is out for sure with their ST-Link. I think some NXP boards might have CMSIS-DAP, but I'm not sure. If you know any board under $20, I'd like to hear about this.

The only thing I found so far that might create a problem is that Atmel EDBG requires full 512 byte payload padded with dummy bytes of the actual payload is smaller. And it sends 512 byte responses. I don't think it is a part of the specification or original ARM CMSIS-DAP code. So I'm curios to see id this tool works with other implementations.

Yes, target specific code will have to be written again for each new target device type, but that usually takes just a few hours. I'm against using flash loaders that other tools use.

ataradov, 2015-09-19:
Also, the tool itself knows nothing about its target. The link you've provided has the code equivalent to mine, not the embedded code.

Notice that there is a file named  flash_algo_gen.py in the link you've provided. This file generates a blob that does exactly the same thing as my algorithm, but it has universal interface to the tool (which is generally slower that the native version, but that may differ depending on the embedded flash controller).

If you want to support a new target, you still need to create a new blob. I'd rather see C code and not an array of bytes :)

ataradov, 2015-09-19:
Ah, ok. So they do embed flash algorithms into the firmware, but for programming though USB MSD. They only use CMSIS-DAP for debugging.

aethaniel, 2015-09-21:
I guess we can close this ticket for now.
Just a last remark: it seems Atmel-ICE is also EDBG-like/CMSIS-DAP v1 (same VID but PID 2141 instead of 2111). The CMSIS-DAP version that ARM have distributed in github is version 2.

closed 2015-09-21 by aethaniel

ataradov, 2015-09-21:
Yes, I've used this tool with Atmel-ICE, it uses exact same chip. It also works with Arduino Zero, for the same reason.

The correct way to recognize CMSIS-DAP devices is not by VID/PID, but by the sub-string "CMSIS-DAP" anywhere in the device name string.

I'm working on another project that is based on this tool, and once I'm done, I'll port all the changes back here.

aethaniel, 2015-11-25:
Hi @ataradov,

Sorry, I missed answering some questions you posted a while ago:

>I eventually want to try other tools, but I don't know any cheap boards that 100% have CMSIS-DAP debugger. ST is out for sure with their ST-Link. I think some NXP boards might have CMSIS-DAP, but I'm not sure. If you know any board under $20, I'd like to hear about this.

It is not clear if last ST-Link firmware for Nucleo board is CMSIS-DAP, but you could find under 20$ these ones:
- Freescale Kinetis KL25 Freedom here: http://www.element14.com/community/docs/DOC-46626
- NXP LPC812: http://www.element14.com/community/docs/DOC-67551

> The only thing I found so far that might create a problem is that Atmel EDBG requires full 512 byte payload padded with dummy bytes of the actual payload is smaller. And it sends 512 byte responses. [...]

Byte payload may differ from one CMSIS-DAP implementation to another. For example Atmel mEDBG (PID:2145) present on Xplained minis has a 64bytes payload. This is the origin of avrdude patch http://savannah.nongnu.org/patch/index.php?8591, following this one: http://savannah.nongnu.org/patch/?8435

ataradov, 2015-11-25:
I've actually got LPC-LinkII and IBDAP (was on sale for $15). Both support CMSIS-DAP. LPC-LinkII required firmware update, but that was straightforward. They both a are detected and work properly. IBDAP uses 512 bytes reports, and LPC-LinkII uses 1024 byte reports, so so far it is the best tool of all of them.

I know that it differs. I've implemented auto-detection for my other projects. CMSIS-DAP command set actually includes a command to get report size, but there is no way to send that command unless you already know report size :)

ataradov, 2015-11-25:
I would like to get ST board with CMSIS-DAP, but so far I can't find any confirmation that one exists.

ataradov, 2015-11-25:
Also, can you confirm that KL25 comes with OpenSDAv2? Because V1 is some proprietary protocol.

aethaniel, 2015-11-25:
about KL25, I can show you only that: https://github.com/mbedmicro/CMSIS-DAP/tree/master/interface/target/hal/DBG_Freescale as I'm not really following what is delivered on non-Atmel boards.

aethaniel, 2015-11-25:
I have read that last ST-Link firmware should be CMSIS-DAP compatible, in relation with mbed partnership for Nucleo boards. I would say obtaining one of the last Nucleo given for free almost everywhere should make the deal. I can ship one to you also.

ataradov, 2015-11-25:
I have one of newer Nucleo boards and it has plain old ST-Link firmware. I don't see any instruction on how to update it to CMSIS-DAP firmware, and as far as I call tell it is impossible.

aethaniel, 2015-11-25:
Look for latest version stlink firmware on web. It should make the deal.