ataradov/edbg #9: releases

aethaniel, 2015-10-21:
Hi @ataradov, do you think you could provide binary releases for all supported OS, please?

I'm integrating this tool into my Arduino SAM port (https://github.com/aethaniel/ExperimentalCore-sam) and I'm able to compile only win64 and linux64 binaries.

Usual OS targets are:
- Win32
- Win64
- Lin32
- Lin64
- OS X

This isn't a strong request but more a nice to have. No stress ;-)

If you have a solution to remote build the project similar to the Travis-CI I'm using at the moment, I could investigate myself and provide you the binaries to bring here for a Release.

Thanks,

T.

ataradov, 2015-10-21:
That's not going to happen:
1. I don't own or have access to any Apple product.
2. I only have Linux64 machine at home.
3. I only have Win64 machine at work.

Creating binaries for a release under those conditions would be a huge nightmare.

I don't want to have them in Git repo, but I can host them on my site if you'll find a way to build all of them.

aethaniel, 2015-10-21:
Ok. Check github release as you don't to know this service.

I will try to do something on my side

aethaniel, 2015-10-22:
you can see here how to do binary release in relation or not with a specific git tag: https://github.com/ataradov/edbg/releases
some complementary information can be found here:
https://github.com/blog/1547-release-your-software
https://help.github.com/articles/creating-releases/

This means you can have your repo with project code, tag this code at some point, release a specific version according to the tag and then associated binaries.

I have like you lin64 and win64 at work and win32 at home on an old computer. A friend has a mac and also a lin32 vm. We can find a way to help you providing these binaries if you want.

ataradov, 2015-10-22:
I really don't see the point of doing releases for this project. The project is very small and there is not active development going on. Just pick the latest master whenever you need something new. The master here is always in a working state.

closed 2015-11-06 by aethaniel