ataradov/edbg #14: Compile for OSX.

poppe34, 2015-12-21:
inside dbg_mac.c I got an error on compile the variable report_size was not declared.  
```c
  char cmd = data[0];
  int res;

  memset(hid_buffer, 0xff, report_size + 1);

  hid_buffer[0] = 0x00; // Report ID
```

I added the line of code:

```c
  char cmd = data[0];
  int res;
  int report_size = dbg_get_report_size();
	
  memset(hid_buffer, 0xff, report_size + 1);

  hid_buffer[0] = 0x00; // Report ID

```

This may or may not be right but it compiled and I am going to make sure it works now.

ataradov, 2015-12-21:
I've added a correct fix. Please test if that works.

I have no MAC, so no way to test.

poppe34, 2015-12-22:
Thanks for the super fast response.  I will start to use this more since it is super convenient.  I will let you know when/if I run into any issues.  I am getting an error but I am trying to verify if it is me or the loader.  Just to give you a heads up here is the error, but again I will verify if it is me.  I am trying to just upload a .bin file so even if (in my mind) the code is dead wrong it should load.

*Error: invalid response while writing the register 0x01 (count = 1, value = 4)*

ataradov, 2015-12-22:
What MCU are you using?

It looks like it fails on this line "dap_write_reg(SWD_AP_CSW, 0x23000052);". Error (value) 4 is a generic fault error, so it is hard to say what is wrong exactly.

poppe34, 2015-12-22:
I am using the Same70 Xplained board.  I had to find alternate programmer tools since it isn't in OPENOCD yet(and on a MAC), and I didn't want to spend the time porting it.

ataradov, 2015-12-22:
I don't have E70 Xplained. I've just tried with V71 Xplained and it seems to work. I've added a patch to set GPNVM boot mode bit. It will not do anything for your problem, but try to short the ERASE jumper and power cycle the board. This will erase everything including fuses. See if this helps with the error. And if it does, updated  version will set that fuse back.

poppe34, 2015-12-22:
No problem.  I wasn't sold that it was the ebdg issue.  I wouldn't do anything about it for now.  If i dig deeper and see an issue I will comment again.

poppe34, 2015-12-22:
FYI... Worked fine after the erase.  Must have hung up on something.

poppe34, 2015-12-22:
Sorry quick question does the atmel edbg chip support SWO?

ataradov, 2015-12-22:
It is possible that device was locked. That may cause all sorts of strange problems.

ataradov, 2015-12-22:
I have no idea about SWO support. I did not get to that stuff yet.

poppe34, 2015-12-22:
I would be happy to help out with SWO if the board debugger supports it.  This a very lightweight scaled down program so you may not want it.  Again great job

ataradov, 2015-12-22:
There is no reason for supporting SWO for this project, this is not a debugger. But I'm working on some other tools, which eventually will need SWO, but not it any sort of short term.

AloyseTech, 2016-01-12:
Hi,

I tried to compile the tool on OSX 10.9 and I get the following error when running "make all" :
[error127.txt](https://github.com/ataradov/edbg/files/87721/error127.txt)

Could you please help to resolve this issue ?

Thanks :smiley:

ataradov, 2016-01-12:
It looks like you are missing GNU autoconf tools. I have no idea how to install them on MAC, but google might help you.

AloyseTech, 2016-01-12:
Thanks for the tip!

here are the command used to install dependencies : 

curl -OL http://ftpmirror.gnu.org/autoconf/autoconf-2.69.tar.gz
tar -xzf autoconf-2.69.tar.gz 
cd autoconf-2.69
./configure && make && sudo make install
 
curl -OL http://ftpmirror.gnu.org/automake/automake-1.14.tar.gz
tar -xzf automake-1.14.tar.gz
cd automake-1.14
./configure && make && sudo make install
 
curl -OL http://ftpmirror.gnu.org/libtool/libtool-2.4.2.tar.gz
tar -xzf libtool-2.4.2.tar.gz
cd libtool-2.4.2
./configure && make && sudo make install

it works now. :smiley:

closed 2016-03-15 by ataradov