ataradov/edbg #16: Does not make all in Windows

FORETRES, 2016-03-15:
Hi, I have followed your instructions for Windows. I inserted the command make all, in the cmd.exe but the following message appears (see message in my Atmel community post - http://community.atmel.com/forum/erase-memory-saml21-whilst-sleep-mode#comment-1825186).

How would I get around this? Thanks

ataradov, 2016-03-15:
I'm not reading AVRFreaks anymore. It appears that you don't have compiler installed. What kind of setup do you have? What happens if you just run "gcc"?

FORETRES, 2016-03-15:
When I run "gcc" it says that it is not recognized as an internal or external command, operable program or batch file. I haven't installed gcc so that might be it. I will do it now and see. Thanks.

ataradov, 2016-03-15:
You need to install MinGW.

FORETRES, 2016-03-15:
Oh yes, I have already installed that to be able to use the make command. Is that only what I need?

ataradov, 2016-03-16:
You need to have GCC in your system PATH variable. 

You should see "gcc: fatal error: no input files" when you run gcc without parameters.

FORETRES, 2016-03-16:
Hi ataradov. Thanks for the help. I have been able to fix my initial problem. Thanks!

closed 2016-03-16 by FORETRES