ataradov/edbg #18: Error: unable to find parent usb device

aswinvenu, 2016-04-07:
Downloads/edbg-master$ make all
gcc -W -Wall -O2 -std=gnu99 dap.c edbg.c target.c target_atmel_cm0p.c target_atmel_cm3.c target_atmel_cm4.c target_atmel_cm7.c dbg_lin.c -ludev -o edbg
aswin@aswin-fourthfrontier:~/Downloads/edbg-master$ ./edbg -bev -t atmel_cm4
Error: unable to find parent usb device

ataradov, 2016-04-07:
What hardware/OS this is? What does lsusb say? What does dmesg say when you plug in the debugger?

aswinvenu, 2016-04-07:
thank you for your faster reply:
results : lsusb
===========
aswin@aswin-fourthfrontier:~$ lsusb
Bus 001 Device 005: ID 0bda:0129 Realtek Semiconductor Corp. RTS5129 Card Reader Controller
Bus 001 Device 004: ID 0cf3:0036 Atheros Communications, Inc. 
Bus 001 Device 003: ID 0c45:670b Microdia 
Bus 001 Device 002: ID 8087:8000 Intel Corp. 
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 003 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 002 Device 002: ID 03eb:2111 Atmel Corp. 
Bus 002 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub

results dmesg when device plugged in:
==============================
[  809.171466] usb 2-1: USB disconnect, device number 2
[  822.126910] usb 2-1: new high-speed USB device number 3 using xhci_hcd
[  822.255395] usb 2-1: config 1 interface 2 altsetting 0 bulk endpoint 0x84 has invalid maxpacket 64
[  822.255404] usb 2-1: config 1 interface 2 altsetting 0 bulk endpoint 0x5 has invalid maxpacket 64
[  822.255817] usb 2-1: New USB device found, idVendor=03eb, idProduct=2111
[  822.255824] usb 2-1: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[  822.255829] usb 2-1: Product: EDBG CMSIS-DAP
[  822.255832] usb 2-1: Manufacturer: Atmel Corp.
[  822.255836] usb 2-1: SerialNumber: ATML2393031800001029
[  822.256087] usb 2-1: ep 0x87 - rounding interval to 128 microframes, ep desc says 255 microframes
[  822.256098] usb 2-1: ep 0x6 - rounding interval to 128 microframes, ep desc says 255 microframes
[  822.257965] hid-generic 0003:03EB:2111.0003: hiddev0,hidraw0: USB HID v1.11 Device [Atmel Corp. EDBG CMSIS-DAP] on usb-0000:00:14.0-1/input0
[  822.258142] cdc_acm 2-1:1.1: ttyACM0: USB ACM device

thank you,

Aswin

ataradov, 2016-04-07:
That all looks good. I've never seen this error before, so I don't really know what may be wrong.

The only thing I can suggest is to run as root and see if that makes any difference.

aswinvenu, 2016-04-09:
No, the same. I am using Debian 8 Jessie. But the same thing happened with ubuntu 14.04 as well(On the same machine). The application works fine in my other two laptops. I doubt there is some hardware related issues. Can you suggest a way to debug the code from hardware level? 

Is there anything to be changed on boot setup?  I am using "Dell Vostro 14 3000 Series" 

Thank you,

Aswin

ataradov, 2016-04-09:
I really have no idea. I just used a standard Linux demo code as a base for the USB stuff , I have no deep knowledge of what it is doing under the hood and how it interacts with the OS. And without a way to reproduce, there is not much I can do to debug this.

aswinvenu, 2016-04-09:
Okay, Thank you. :)

ataradov, 2016-04-09:
If you want to spend some time debugging this, then we can try.

For starters, I'd print out the value of 'path' in the "udev_list_entry_foreach(dev_list_entry, devices)" loop right after the line"path = udev_list_entry_get_name(dev_list_entry);".

This will show if it found more than one USB HID device and some of them are not behaving properly.

Alternatively, try to run the program when no debuggers are attached. If it fails the same way, then there is some weird device in the system and path printout will help to understand which one.

ataradov, 2016-04-09:
And if previous test shows that this is a problem, then simply change "check(parent, "unable to find parent usb device");" to "if (NULL == parent) continue;" to skip than device.

aswinvenu, 2016-04-09:
Yes you are right. I am getting the same error when the debugger is unplugged as well. this is the printed output /sys/devices/pci0000:00/INT33C3:00/i2c-1/i2c-DLL0662:00/0018:06CB:2985.0001/hidr

aswinvenu, 2016-04-09:
Yes.... We did it !!! Awesome !!! It works.

Thank you so much :+1:

ataradov, 2016-04-09:
This is an I2C touchpad from  Synaptics pretending to be a HID device. But it is not a part of the USB tree, obviously, so it has no parent. I'll change the code accordingly.

aswinvenu, 2016-04-09:
Alex I am adding you to my linkedin network please accept my request. I need to learn lot from you.

ataradov, 2016-04-09:
Code updated.

closed 2016-04-27 by ataradov