ataradov/edbg #21: Support for ATSAMB11

arsinio, 2016-05-11:
Awesome work so far! Very happy to see some options opening up for development outside of AVR Studio...

Anyways, more of a feature request than an issue...I have an ATSAMB11 development kit and I'm getting the following error when trying to program:

`Debugger: ATMEL EDBG CMSIS-DAP ATML2534030200001457 03.0C.0191 (S)
Error: unknown target device (CHIP_ID = 0x00000000)`

I've tried with all of the target options (no success)...Any ideas?

ataradov, 2016-05-11:
SAMB11 is an ARM Cortex-M0 (not +) device that came with NMI acquisition, it has completely different programming algorithm, so it is not supported by the EDBG at the moment.  There is also no publicly available documentation that would allow me to create a programmer.

arsinio, 2016-05-11:
By "supported by the EDBG" I'm assuming you mean this project?

The SAMB11 XPlained Pro dev kit definitely has an EDBG on it that is used to program the B11...but not much help if we don't have any documentation to go on...

ataradov, 2016-05-11:
Yes, i meant this project. The chip EDBG supports this device, as it is a standard SWD programmer. This is evidenced by "CHIP_ID = 0x00000000". This value comes from a known memory location that is the same for all Cortex-M0+ devices. Obviously the same location on SAMB11 contains 0.

closed 2016-05-20 by ataradov