ataradov/edbg #23: Problem while reseting SAM4L

euripedesrocha, 2016-08-20:
I'm trying to use edbg to program, and later debug, a SAM4LXplained pro board.

I had modified target.c to add the uC part number( ATSAM4LC4CA) but I'm not being able to execute any comand.
I've tried:
``` bash
edbg -be -t atmel_cm4
```
And had the following error:

Debugger: ATMEL EDBG CMSIS-DAP ATML1783020200003186 03.19.01A3 (S)
Error: invalid response while writing the register 0x0d (count = 1, value = 4)

With gdb I found that the problem occur when the dap command to AIRCR for reset is being written. 

I think that is possibly not a problem with edbg, but I want some help to figure out a solution, and to add it to edbg

ataradov, 2016-08-20:
I'm on vacation at the moment with no access to any hardware. I'll be back in a couple of weeks and I'll have a look at this.

loiclefort, 2016-08-22:
@euripedesrocha, I don't know about the AIRCR write error, but adding the SAM4L device id to target_atmel_cm4 is not enough anyway.  target_atmel_cm4 only supports the flash controller that is present in other SAM4 devices (EEFC) and SAM4L has a completely different flash controller (FLASHCALW).

euripedesrocha, 2016-08-22:
@loiclefort Thanks for the info about the flash controller.

The reset error seems to be hardware related, or some configuration missing(clock frequency for comm?). Also I discovered that the board needs to be powered through both USB connectors it has. At first I was only using the debug connector. After this I used openocd to write the flash. edbg still can't reset the target.

I'll try to put some effort again in few weeks.

closed 2020-11-21 by ataradov