ataradov/edbg #24: Verification Error on SAM E70

hoppel, 2016-08-30:
Hello, 
I tried to flash an SAM E70 on the Atmel SAME7 Xplained Dev Board.
I compiled edbg from scratch. libudev-dev was need, so I installed this via apt-get. 
The build had no problems. 

When I tried to flash the binary, I got this error:

```
~/workspace-Qdev/SAME70-Blinky $ edbg -bpv -t atmel_cm7 -f bin/SAME70-Template.bin 
Debugger: ATMEL EDBG CMSIS-DAP ATML2637010000000000 02.09.0169 (S)
Target: SAM E70Q21
Programming........,....................................... done.
Verification...................................
at address 0x404000 expected 0x86, read 0x00
Error: verification failed
```
I seems that the MCU is not programmed. I am not sure, where I have to look
to find the problem. 

Regards,
Andre

hoppel, 2016-08-30:
Oh, that was quick. I must erase the chip before. 
```
 edbg -ebpv -t atmel_cm7 -f bin/SAME70-Template.bin 
Debugger: ATMEL EDBG CMSIS-DAP ATML2637010000000000 02.09.0169 (S)
Target: SAM E70Q21
Erasing...  done.
Programming........,....................................... done.
Verification.......................................... done.
```
Maybe you should add the argument to your example in your README.md

ataradov, 2016-08-30:
It should work without full chip erase. Those dots before the comma after "Programming..." is erase of relevant sectors (only things that are needed to program the binary). 

This may be a bug, I'll verify this next week, I have no access to any hardware at the moment.

ataradov, 2016-09-08:
This issue is now fixed.

closed 2016-09-08 by ataradov