ataradov/edbg #27: Cannot run make all command on mac

shahbhaumik, 2016-11-08:
After cloning, when I try to make it throws error :

git clone git://github.com/signal11/hidapi.git
Cloning into 'hidapi'...
fatal: unable to connect to github.com:
github.com[0: 192.30.253.113]: errno=Operation timed out
github.com[1: 192.30.253.112]: errno=Operation timed out

Any ideas why? I even tried it as a super user, same issues.

ataradov, 2016-11-08:
It tries to clone this repo https://github.com/signal11/hidapi and fails for some reason. Try to clone it manually and remove that line from Makefile.

MAC OS stuff was contributed, and I don't have any Apple products, so I can't really check if MAC OS port is working.

shahbhaumik, 2016-11-08:
Yes,
It worked, after couple of correction the make file, I had to give it super user rights to do few things.

I was able to install it just fine, later will test if it can flash the SAM micro.

Thanks for your quick response.

closed 2016-11-08 by shahbhaumik