ataradov/edbg #31: Supported file format

iDoka, 2017-03-17:
Hi, Alex!

I'm linux user and edbg is great thing for CLI project flow!

Which one file format supported by your tool?
bin/hex/elf???

I think edbg need simlpe check for file-extension for preventing brick the uC-device.

WBR,
Dmitry

ataradov, 2017-03-17:
The only supported format is raw binary. I don't want to limit file extensions, since not all programmed images are actually application code. You can program data sections as well. And some compilers like to produce files with wired names by default.

ataradov, 2017-03-17:
Also, there is no way to fully brick the device. You can always just do the right thing again. 

If you are afraid of overwriting some sections, for which you don't have binary files, then it is a good idea to read the whole device first anyway, regardless of the used tool.

closed 2017-03-27 by ataradov