ataradov/edbg #37: Device stuck with DAP_TRANSFER_WAIT return code

dwinkler2, 2017-09-11:
I have several samr21xpro devices each attached to several machines running Debian Jessie. Edbg has been flawlessly loading binaries to all of these devices for some time.

The latest code that was loaded for debugging onto the devices seems to have caused many (perhaps 2/3) of the samr21xpro to no longer work with edbg. All edbg operations now fail with the following: 
"Error: invalid response while writing the register 0x00 (count = 1, value = 2)"

Looking through dap.c, count=1 seems to be the expected output, but value=2 seems to correspond to the DAP_TRANSFER_WAIT return code. 

My understanding is that similar issues can arise when the security bit is set (I don't think I've accidentally done this), but that a chip erase will undo even the worst goof-ups. However, in this case even the erase operation fails in the above manner.

ataradov, 2017-09-11:
It looks like those devices are in the sleep mode. You can try to reset those devices manually fast and at the same time retry the command (probably just erase).

Security bit will prevent any access at all. In this case debug subsystem is fully accessible, it jist can't do anything with the core, since it is unresponsive.

dwinkler2, 2017-09-11:
Thank you for your quick reply. This seems strange, as a bug in the firmware causing the samr21 to remain in sleep mode could potentially brick the device without manual resets? 

I will give it a try and see if it resolves my problem, but it seems to be unrelated to edbg so I will close the issue. Thank you again for your help.

closed 2017-09-11 by dwinkler2

ataradov, 2017-09-11:
It is not permanently bricked, but yes, having device (any architecture in my experience) sleep, makes it harder to reprogram.

Debugger resets the device and keeps it in reset, but in order to do that it has to get the core (buses actually) in the running state.

I always put ~1 second dummy loop in my sleeping programs when debugging. This lets debugger intercept control before application goes into sleep.