ataradov/edbg #38: MinGW shows multiple EDBG debugger attached

dylad, 2017-09-13:
Hi,
I'm trying to use edbg on MinGW. I used a SAMD20-Xplained board to program a custom board based on SAML21.
When I used edbg on linux or linux vm, everything runs fine.
But on MinGW, edbg said there are several debuggers attached (which is not the case)
here the output of `edbg --list`
`Attached debuggers:
  ATML1873040200013344 - Atmel Corp. EDBG CMSIS-DAP
  EDBG CMSIS-DAP - EDBG CMSIS-DAP EDBG CMSIS-DAP
  EDBG CMSIS-DAP - EDBG CMSIS-DAP EDBG CMSIS-DAP
`
If I specify `edbg -s ATML1873040200013344`, it works as expected.
This is no a main issue but this is a bit annoying.
Any ideas ?

ataradov, 2017-09-13:
This is really bizarre, I've never seen anything like this. Can you plug this board into a Linux machine and show what apeears in the dmesg output?

dylad, 2017-09-13:
Here the dmesg output from a linux VM : 

[ 1345.598645] usb 1-2: new full-speed USB device number 7 using ohci-pci
[ 1345.827291] usb 1-2: New USB device found, idVendor=03eb, idProduct=2111
[ 1345.827295] usb 1-2: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[ 1345.827297] usb 1-2: Product: EDBG CMSIS-DAP
[ 1345.827299] usb 1-2: Manufacturer: Atmel Corp.
[ 1345.827300] usb 1-2: SerialNumber: ATML1873040200013344
[ 1345.872135] hid-generic 0003:03EB:2111.0005: hiddev0,hidraw1: USB HID v1.11 Device [Atmel Corp. EDBG CMSIS-DAP] on usb-0000:00:06.0-2/input0
[ 1345.872208] cdc_acm 1-2:1.1: ttyACM0: USB ACM device

Regarding the D20 board this is a SAMD20 Xplained Pro based on SAMD20J18A. Atmel Studio gives me the following infos : 
Device Signature : 0x10001100
Revision : B

ataradov, 2017-09-13:
This looks good. I think this is some other device in your system that makes it fail this way.

The quick fix that may work: in dbg_win.c:

    if (INVALID_HANDLE_VALUE != handle)
    {
      wchar_t wstr[MAX_STRING_SIZE];
      char str[MAX_STRING_SIZE];

      wstr[0] = 0; // <- add this
      str[0] = 0; // <- add this
........

Later today I'll add checking for return value of HidD_GetXXXString(), which is probably a better fix.

ataradov, 2017-09-13:
BTW, how do you even make D20 Xpro it use external connector?

dylad, 2017-09-14:
Thanks for handling the problem ! I'll try any patches and let you know.
I use this [little adaptor](https://www.olimex.com/Products/ARM/JTAG/ARM-JTAG-20-10/) from Olimex which is very convenient and affordable. (also available on other websites)
Then I connected Vccref,GND,nRST,SWDIO,SWCLK to my board. At power-up, the D20 board detects the 3,3V from my own board and switch to it (there a LED near to the small debug connector on the D20 indicating the board switches to an external device).

dylad, 2017-09-14:
I don't know if this can help you but when I connected a SAML21-Xpro board to my computer, there is only one entry in `edbg -l` but when I connected my SAMD20-Xpro alone (without the Olimex adaptor and my custom board) there are several entries in `edbg -l`.

ataradov, 2017-09-14:
Can you please check the version of the EDBG firmware on the D20 board? Just run
`edbg -b -s ATML1873040200009875 -t atmel_cm0p` (replace your SN, of course)
You should see something like this:
`Debugger: ATMEL EDBG CMSIS-DAP ATML1873040200009875 01.0F.00DB (S)
Target: SAM D20J18A`. And `01.0F.00DB` is version here. 

Apparently it detects external target only on reset, not in run-time. That was my mistake. I can now make both ports work fine on my hardware.

ataradov, 2017-09-14:
Also, did the fix with setting first character of the strings to 0 do anything on your system? I'm considering it for a permanent solution.

dylad, 2017-09-14:
Here the requested terminal output : 
edbg -s  ATML1873040200013344 -t atmel_cm0p -b
Debugger: ATMEL EDBG CMSIS-DAP ATML1873040200013344 03.22.01B3 (S)
Target: SAM L21J18A

I just test your fix -> it works like a charm ! 
Thanks a lot @ataradov !

ataradov, 2017-09-14:
That's L22, I was looking for the same output from the D20 board. They have different version of the firmware for different boards.

I'll commit this fix later today, then.

dylad, 2017-09-14:
No it was my D20 board but I forgot to remove my custom board.
New terminal output (which is the same): 
```
edbg -s  ATML1873040200013344 -t atmel_cm0p -b
Debugger: ATMEL EDBG CMSIS-DAP ATML1873040200013344 03.22.01B3 (S)
Target: SAM D20J18A
```
Thanks for the incoming patch !

ataradov, 2017-09-14:
I've updated to the same version, and still no problems on my side. 

I assume you have some other HID devices attached to the system that don't supply identification strings for whatever reason.

dylad, 2017-09-14:
The problem is definitely on my system. Thanks for helping me with this.
I'll close this issue once your fix will be merged.

ataradov, 2017-09-15:
Done.

dylad, 2017-09-15:
Good job, thanks for your time :)
I'm closing the issue.

closed 2017-09-15 by dylad