ataradov/edbg #40: SAMD21G18

criticalboot, 2017-09-18:
Hi
I am using samd21G18A chip, with atmel ice I am trying to burn bootloader to it.  I am getting the following response when using the command. 
`./edbg -t atmel_cm0p`
`Error: invalid response while writing the register 0x00 (count = 0, value = 7)`

what can be the cause for the error.

ataradov, 2017-09-18:
This can be a lot of thing, starting from a dead device or broken connections.

criticalboot, 2017-09-19:
Sorry, my bad. broken connections.

after fixing connections I am getting

```
./edbg -e -t atmel_cm0p
Error: unknown target device (DSU_DID = 0x10010306)

```
can you throw some light on this?

ataradov, 2017-09-19:
You need to add that device in the table of devices in target_atmel_cm0p.c.
Just based on that ID, it would read:

`  { 0x10010306, "SAM D21G18A (Rev E)", 256*1024, 4096 },`

criticalboot, 2017-09-19:
Thanks, for your help. Now able to upload the program.

closed 2017-09-19 by criticalboot