ataradov/edbg #47: Windows dependencies

ooxi, 2018-01-27:
Hi,

first of all, thank you very much for this great project! I have a question regarding Windows dependencies. The readme states that no dependencies on windows are required.

However [dbg_win.c](https://github.com/ataradov/edbg/blob/master/dbg_win.c#L34) seems to need the Windows Driver Kit? I'm asking because I'm currently trying to cross compile edbg for Windows and I'm missing the `ddk` include directory.

Maybe the README should extended by stating the Windows Driver Kit is required, but that's normally bundled with Visual Studio?

-- ooxi

ataradov, 2018-01-27:
MinGW comes with all the headers you need, and I don't really care what is bundled with VS. 
I'm not interested in supporting VS at all.

ooxi, 2018-01-27:
Thanks for this information, then I'm most likely just missing a download :-)

closed 2018-01-27 by ooxi

ataradov, 2018-01-27:
Earlier version of MinGW put hidsdi.h and hidpi.h in different locations, but it looks like they are now in "ddk". Check if you already have those files somewhere in MinGW includes.