ataradov/edbg #57: Provide encrypted bintray API key

ooxi, 2018-02-28:
# bintray

If you are logged into bintray, you should see a list of repositories on the left named *Owned Repositories*. There you can *Add New Repository*.

Please create a public repository named *edbg* of type *Generic*. Inside the repository you need to add a package named *travis-ci*. You can of cause change the names to whatever you like, I will change my pull request accordingly.

Now with the repository and package being created, we need to extract your bintray API key. It's available in the bintray web GUI at [*Edit Your Profile*](https://bintray.com/profile/edit) / *API Key*

Please do not share your API key here directly, but encrypt it using Travis CI first. The rest of this document assumes your API key is `super_secret`. In reallity it should look like a long hexdecimal character sequence.

# Travis CI

[Encrypt](https://docs.travis-ci.com/user/environment-variables/#Encrypting-environment-variables) your bintray credentials for usage in Travis CI

```
travis encrypt -r ataradov/edbg BINTRAY_DEPLOYMENT_API_KEY=super_secret
```

You might need to install the `travis` gem, I recommend doing so inside an ephemeral container:

```
$ docker run -it --rm ubuntu:16.04
# apt-get -y update && apt-get -y install build-essential ruby ruby-dev
# gem install travis
# travis encrypt -r ataradov/edbg BINTRAY_DEPLOYMENT_API_KEY=super_secret
```

In my opinion, encrypting secrets for usage inside Travis CI is the most annoying part. Travis CI should offer a GUI on their website for this task ☹

You should see an output similar to

```
  secure: "mMXpe5Vhl6vgvT/D9HwfPgamDJsuk4q9cKRidIqp3Xe2TmOr5PZ3KMxXaJhNBzRPiI/kH2pYImK3EG2pbAhFUj2D3YKyyB6B1FrE7YQB72MzTFn86MY1L+igywap/3dm7ZRDT0pGLSQCjVN9QOpbKzpcCFlLu0oLMFT23FNOj6BFN/+fHJrIW9b4x44vd3P2MPk2E4CYMr43seedPzUHmio4mrqXJKAa3sHH9BvS/E6tpeYiWNJ1+RO2e/p5AqrS1d157+hnH8upsC2BIwlkRbUEkZdrjsJjc+CNVS2IBPjXbfeD+nGIJdiKP33+s3i1lnyMUAmy8hmZ80X2sKTFElow0cxXVhEI8uwLmylSmMR50VpmLHM2VZJV16dNJ2HvadDm4ztm82fPL4Ob7sybDK34+of8BYXoSWDBJ6Fx0ABISQ1z7riargPxdGI09a0+z6ZbJwARWWPD0K0av/W/PpCMTSnV4zmcWilTkw5t9JhIRLXral5Z/ghS7u1qVBvS6SgSf46e41oSAgRJyfnPl21R4D8uU6/b2MqvslQ0re+JKMdcrSROCYIc/455SwTol1Q0YAm5d7kCFSKVVS+dfV3BdtcxvgO57VcbSRsppTSC+PzZPh0EGgf2KUyb2Rj+ZltaqJYiWgpUyVdizrfE0/kqEkE17G7lj+wh3TpoAk4="
```

Please paste this string here. While your bintray API key is secret and must not be shared with anybody, the encrypted string is not. It's encrypted with your public Travis CI key, thus only Travis CI will be able to decrypt the string and inject additional secret environment variables into your build environment.

ataradov, 2018-03-02:
All repos and packages are named as described. The package has a mandatory field called "Version Control" with no description, so I just put GitHub URL there.

ooxi, 2018-03-02:
Thanks! I will prepare a pull request

closed 2018-03-02 by ooxi