ataradov/edbg #58: Review bintray directory structure

ooxi, 2018-02-28:
Hi,

as discussed in pull request #56, I have set up a [proof of concept](https://github.com/ooxi/edbg/blob/private/bintray/.travis.yml#L70) of how to automatically deploy edbg binaries to bintray using Travis CI.

The current setup will result in an URL looking something like

> https://dl.bintray.com/ataradov/edbg/master/

where *master* is the branch being build. This directory contains a list of all revisions build by Travis CI. The binaries inside a build are currently named like this:

```
edbg-b36-LINUX_CLANG-1657458
edbg-b36-LINUX_GCC-1657458
edbg-b36-OSX_CLANG-1657458
edbg-b36-OSX_GCC-1657458
edbg-b36-WINDOWS-1657458.exe
```

* *36* is the Travis CI build number
* *LINUX_CLANG* is the target's name
* *1657458* are the first eight digits of the git revision build by travis, in this case [165745819852853cf625db28878d240be2aa09ea](https://github.com/ooxi/edbg/commit/165745819852853cf625db28878d240be2aa09ea)

I personally dislike the mixing of SNAKE_CASE and kebab-case, so maybe we should change the target names. What do you think?

ataradov, 2018-02-28:
I don't care about the naming at all, anything works for me.
I'll try to find some time to set up bintray later this week. I'm a bit busy at the moment.

ooxi, 2018-02-28:
{quote}
I don't care about the naming at all, anything works for me.
{quote}

Great, I therefore consider this task done

{quote}
I'll try to find some time to set up bintray later this week. I'm a bit busy at the moment.
{quote}

Sure no hurry :-)

closed 2018-02-28 by ooxi