ataradov/edbg #61: Restarted builds fail because bintray does not allow overwriting existing executables

ooxi, 2018-03-03:
I noticed a couple of Travis CI builds look like they failed. The reason is that a restarted build gets the same build number and bintray does not allow overwriting of build artifacts by default.

```
Depolyment to Bintray failed with response {"message":"Unable to upload files: An artifact with the path 'master/19/edbg-b19-OSX_GCC-72ab2d2' already exists"}
```

I created this issue as a reminder for myself. I'll try to find a solution, at least I will check if the file exists before trying to upload it so the build will not be marked as failed in Travis CI.

ooxi, 2018-03-03:
According to the [bintray documentation](https://bintray.com/docs/api/#_upload_content), adding `[?override=0/1]` to the `PUT` request should be sufficient.

I'll try this and prepare a pull request.

ataradov, 2018-03-03:
I made the change myself.

closed 2018-03-03 by ooxi

ataradov, 2018-03-03:
My fix did not really work, no uploads happened. I just tried a different syntax, let's see if that works.

ataradov, 2018-03-03:
Ok, that works, but one of the uploads failed due to bintray issue. That's a good change to try to re-run the job and see if overrides actually work.