ataradov/edbg #71: SAML10/L11 Support

bousqi, 2018-10-22:
Could you consider to add support of last Microchip core Cortex M23, SAML11 family ?

https://www.microchip.com/promo/sam-l10-and-l11-microcontroller-family

ataradov, 2018-10-22:
Yes, I've started work on them some time ago, but they are highly unconventional as far as programming goes.

I will complete the work this week, since now there is an actual request for it.

bousqi, 2018-10-22:
I did performed few tries to add support on my side and so far verify, read, fuses, and erase is working. 
I just have troubles with write.

Many things have changed with M23, and my first shoot does not consider the different DAL.
The device must be in DAL2 with access to NVMCTRL_SEC.

Would you like me to share my "half a day" work ? (not complete at all)

ataradov, 2018-10-22:
I've got that working as well, that's the least of the problems.  The problems come to the proper device initialization and entering into boot ROM PARK mode. Plus probably adding a new command line switches to support erase keys.

bousqi, 2018-10-23:
Totally agree with you... Boot Rom management seems to be tricky. I'll get back to you if I have the opportunity to play with it.

ataradov, 2018-10-24:
I just pushed the changes for SAM L10/L11 support. I only did some basic testing, but it appears to work.

ataradov, 2018-10-24:
To accommodate User Row and BOCOR, I introduced section index in the fuse operations. Now section 0 refers toUser Row and section 1 to BOCOR.

Something similar will be done for data flash, but that will happen much later. For now we can't program data flash.

closed 2018-10-24 by ataradov

bousqi, 2018-10-24:
Seems operational on my side too.
I'll play with it.

ataradov, 2018-10-24:
Yes, and erase key for now is fixed to 0xffff...ffff (default). This will also have to be passed from the command line somehow.