ataradov/edbg #72: Windows build dependencies

bousqi, 2018-10-22:
Building edbg for windows has some dependencies (while none are indicated in README.txt)
dbg_win.c includes the followinf ddk files : 
```
#include <ddk/hidsdi.h>
#include <ddk/hidpi.h>
```

Consequently, building fails with : 
```
$ make
gcc -W -Wall -Wextra -O2 -std=gnu11 dap.c edbg.c target.c target_atmel_cm0p.c target_atmel_cm3.c target_atmel_cm4.c target_atmel_cm7.c target_atmel_cm4v2.c  dbg_win.c -lhid -lsetupapi -o edbg.exe
dbg_win.c:34:10: fatal error: ddk/hidsdi.h: No such file or directory
 #include <ddk/hidsdi.h>
          ^~~~~~~~~~~~~~
compilation terminated.
make: *** [Makefile:45: edbg.exe] Error 1
```

Any chance to for you to add them in your git rather than pull the whole ddk ?

Thanks.

ataradov, 2018-10-22:
They should be included with MinGW. Some versions of MinGW don't have "ddk" directory, so try just  
```
#include <hidsdi.h>
#include <hidpi.h>
```
Otherwise see if your installation of the MinGW is complete.

bousqi, 2018-10-23:
I confirm that your trick works.
I just had to hard set the compiler with 
`    COMPILER = i686-w64-mingw32-gcc`
in windows section of makelfile.

closed 2018-10-24 by ataradov