ataradov/edbg #74: Commit 800e209 causes verification error for SAMD21J18

photonthunder, 2019-02-26:
After commit 800e209 I get verification errors anytime I attempt to program my samd21j18 board:

edbg -t atmel_cm0p -bpv -f boot.bin
Debugger: ATMEL Atmel-ICE CMSIS-DAP J41800049111 01.26.0081 (SJ)
Clock frequency: 16.0 MHz
Target: SAM D21J18A (Rev D)
Programming........................... done.
Verification...
at address 0x0 expected 0x38, read 0xff
Error: verification failed
make: *** [flash] Error 1

I played around in target_atmel_cm0p.c and I could get occasional success by tweaking some of the changes (like converting dap_write_half and dap_write_byte back to dap_write_word) but no one line fix.

ataradov, 2019-02-26:
This is very strange. I don't have D21 boards that have SWD connector exposed. I tried with D21 Xpro and EDBG, and D11 and Atmel-ICE and both things worked.

I will read through the code and try to figure this out. It was somewhat a big change, so errors are possible.

ataradov, 2019-02-26:
Can you read the fuses from your device?

ataradov, 2019-02-26:
Can you also try some experiments to possibly narrow this down. Like program using the old version, and verify using the new one. I feel like the programming that actually fails, but I want to verify.

photonthunder, 2019-02-26:
So if I try setting the EEPROM size fuse I get the following:

edbg -t atmel_cm0p -F wv,6:4,0x04
Error: fuse verification failed: expected 0x4 (4), got 0x7 (7)

Could it be timing between the unlock and the write?

ataradov, 2019-02-26:
This is still not conclusive. It is ether read or write fails. Fuses use essentially the same procedure as normal programming, but fuses are not locked, so it can't be unlock timing issue.

Try to lower the clock a lot. And do write on the old verify on the new test. This will at least tell which part is broken.

I don't see anything wrong with the code, really.

photonthunder, 2019-02-26:
old edbg:

edbg -t atmel_cm0p -bp -f boot.bin
Debugger: ATMEL Atmel-ICE CMSIS-DAP J41800049111 01.26.0081 (SJ)
Clock frequency: 16.0 MHz
Target: SAM D21J18A (Rev D)
Programming........................... done.

latest edbg:
edbg -t atmel_cm0p -bv -f boot.bin
Debugger: ATMEL Atmel-ICE CMSIS-DAP J41800049111 01.26.0081 (SJ)
Clock frequency: 16.0 MHz
Target: SAM D21J18A (Rev D)
Verification........................... done.

ataradov, 2019-02-26:
So the actual write is broken. Try to lower the clock frequency. 

I just tried with multiple D21 device of various revisions and I don't see the problem.

ataradov, 2019-02-26:
Try to replace all dap_write_half() and dap_read_byte() in target_program() with corresponding word functions. You don't need to change anything else (like addresses or bit values), it all should work the same way.

This will eliminate size switching entirely.

ataradov, 2019-02-26:
Also, the code now absolutely relies on reset extension, so you must have reset line from the debugger connected to the MCU.

photonthunder, 2019-02-26:
I was just going to say that slowing the clock down helped and it will complete on occasion, but that seems irrelevant because we do not use the reset line from the debugger in our current board.

ataradov, 2019-02-26:
Then you are missing writes to ARM_DAP_DHCSR and ARM_DAP_DEMCR in the target_select() to stop the core. and corresponding writes in the target_deselect().

ataradov, 2019-02-26:
It should be safe to add them (without the write to ARM_SCB_AIRCR). I will need to think about this.

I switched to using the reset extension, since it is a more reliable way to connect to the MCU, since it can work with device that immediately goes to sleep, for example. And this is also absolutely necessary on SAM L10/L11.

I will experiment with adding this code back for CM0+ targets, and if it work, I will add it back. You can try the same on your hardware.

photonthunder, 2019-02-26:
It was never really an issue in the past so we took it off this particular production board.  We do see programming locks on occasion but we would "fix" by doing an erase in a loop while manually resetting the board.

ataradov, 2019-02-27:
I think just adding "dap_write_word(ARM_DAP_DHCSR, 0xa05f0003);" at the beginning of target_select() will solve your problem. It is technically a correct way to use those device (without a reset), so I will add this code back after I do some more testing.

photonthunder, 2019-02-27:
Thanks for the quick response and the help.  That does appear to fix my issue.  For future builds, would you recommend adding the reset back in?  What are the main benefits?

ataradov, 2019-02-27:
It is not 100% necessary for those device unless toll makers make stupid mistakes like that :). 

It is probably not a big deal for production, so if saving test pads is the goal, then don't put it in. 

But it is very helpful for debugging, since it is possible to put the part into a very hard to recover state. For example if your code immediately goes to  sleep on reset. Normally debugger will not be able to recover the part, unless it catches it at the exact right time, which may be hard.

This system lets the debugger intercept the core before it executes the first instruction.

But I will add the software rest code back later today after some more testing on different targets.

ataradov, 2019-02-27:
It is however mandatory for SAM L10/L11, so if you ever plan to use them, keep that in mind.

photonthunder, 2019-02-27:
I appreciate the tool and the insight.  Thanks for the assistance.

ataradov, 2019-02-27:
Should be fixed by the latest commit.

closed 2019-02-27 by ataradov

photonthunder, 2019-02-27:
Looks like the code addition in target_deselect breaks programming again:

`dap_write_word(AIRCR, AIRCR_VECTKEY | AIRCR_SYSRESETREQ);`

If I take it out, it works fine.

ataradov, 2019-02-27:
How exactly does it break it? This code is executed after the programming is done, so presumably at least one programming cycle succeeds?

This is just a software reset of the MCU.

photonthunder, 2019-02-27:
Looks like it is verification that fails:

./../edbg/edbg -t atmel_cm0p -bpv -f boot.bin
Debugger: ATMEL Atmel-ICE CMSIS-DAP J41800049111 01.26.0081 (SJ)
Clock frequency: 16.0 MHz
Target: SAM D21J18A (Rev D)
Programming........................... done.
Verification...
at address 0x0 expected 0x38, read 0xff
Error: verification failed
make: *** [flash] Error 1

photonthunder, 2019-02-27:
In this case verification succedded but then the fuse I ran didn't:

./../edbg/edbg -t atmel_cm0p -F wv,2:0,0x07
./../edbg/edbg -t atmel_cm0p -bpv -f boot.bin
Debugger: ATMEL Atmel-ICE CMSIS-DAP J41800049111 01.26.0081 (SJ)
Clock frequency: 16.0 MHz
Target: SAM D21J18A (Rev D)
Programming........................... done.
Verification........................... done.
./../edbg/edbg -t atmel_cm0p -F wv,2:0,0x03
Error: fuse verification failed: expected 0x3 (3), got 0x7 (7)
make: *** [flash] Error 1

ataradov, 2019-02-27:
I will experiment with fast execution of multiple commands in a row.

I'm not sure why it would break this way, this reset is the same as it was on the old code.

photonthunder, 2019-02-27:
So it looks like if I add 

`dap_write_word(DEMCR, 0x00000001);`

back to target select then it starts working.

After I run my commands I can then remove the code and the lastest master works as well?  Insight?  Does that register hold through a reset, or because I am not doing a debug reset it doesn't get cleared?

ataradov, 2019-02-27:
Yes, and you would also need to return AIRCR to the 

Writing DEMCR by itself only requests the core halt on reset, but does not actually do anything immediately. Write to AIRCR resets the core that will be stuck on the first instruction.

I guess what happens now is that the core gets halted at a random place after it already executed some code that changed the MCU state and interferes with the debugger.

And setting DEMCR back to 0 in the deselect before writing AIRCR will let the code run.

I will put those instructions back in later today. I want to define all the magic numbers while I'm at it.

photonthunder, 2019-02-27:
Sounds good, thanks

ataradov, 2019-02-28:
I've put all that stuff back. It all should work now.

photonthunder, 2019-02-28:
Working great now, thanks again...