ataradov/edbg #77: Target is reset twice.

jcarrano, 2019-03-07:
## Description

This change (https://github.com/ataradov/edbg/commit/b65dc2435a68bc0ab32412e4826333e2b6da5f42#diff-00259a9e325566089cd936bcc3e75e50R564) caused targets to be reset twice. This is relevant when one just wants to reset (using, e.g., `edbg -t atmel_cm0p`)

The first reset: https://github.com/ataradov/edbg/blob/b65dc2435a68bc0ab32412e4826333e2b6da5f42/edbg.c#L564
The second reset: https://github.com/ataradov/edbg/blob/b65dc2435a68bc0ab32412e4826333e2b6da5f42/edbg.c#L680

### Why is it relevant

In the [RIOT OS](https://github.com/RIOT-OS/RIOT) test suite we reset the boards at the beginning of each test. The test script waits for a known string to start the test. With the current edbg behaviour the test starts and shortly afterwards the board is reset, causing it to fail. Since this is happening only with atmel boards using edbg, I'd like to avoid having to add a specific kludge to handle it in the test infrastructure if I can avoid it.

## Steps to reproduce

Load an program that does something immediately after boot, for example output a string over the UART. In my case I'm running the [RIOT OS](https://github.com/RIOT-OS/RIOT) test suite, and listening on the serial connection.

Reset the board with `edbg -t <target type>` (in my case `atmel_cm0p`.)

### Expected result

Board is reset once thus the message is sent once.

### Actual result

Board is reset twice in a short span thus the message is sent twice.

ataradov, 2019-03-07:
It is probably safe to switch to "dap_reset_target_hw(0);"  CM4v2 at this point does a reset with extension in the target_select() code anyway.

The next problem would be that target_deselect() does a software reset, which would be followed by the hardware reset. This is done to ensure operation without a reset pin.

I have no problem with the first change, in fact I think the whole reset may be removed from here and moved to individual targets select procedure. I will do that later today after some testing.

But the second one will be remain as is, so if this is still a problem, removing the first reset will not help much.

It is possible that target_deselect() and dap_reset_target_hw(1) happen fast enough that MCU does not have time to run any useful code, of course.

jcarrano, 2019-03-07:
@ataradov Thanks for the quick response.

> It is possible that target_deselect() and dap_reset_target_hw(1) happen fast enough that MCU does not have time to run any useful code, of course.

Yes, I think that is the case, because I don't experience any issue with commits prior to b65dc24.

ataradov, 2019-03-08:
I've moved the initial reset into the target specific code. This should fix your issue.

closed 2019-03-08 by ataradov

jcarrano, 2019-03-11:
I can confirm that the issue is solved. Thanks for the prompt response!