ataradov/edbg #78: Error: invalid response while writing the register 0x00 (count = 0, value = 0) on ATSAMD51J20A using atmel-ice

NoeelMoeskops, 2019-03-20:
Hi,

I'm trying to connect to an ATSAMD51J20A using an atmel-ice debugger. command: 
```
> sudo ./edbg-b56-LINUX_GCC-4f5d490 -t atmel_cm4v2 -c 2000

Error: invalid response while writing the register 0x00 (count = 0, value = 0)
```
`sudo` is necessary because of permission issues.
All the light flash on the debugger so I think that it is connecting properly. Whenever I disconnect the MCU I get `(count = 0, value = 7)`, so something is going on.

Do you know how to solve this issue?

ataradov, 2019-03-20:
What is the hardware and pins connected to the debugger?

One of the reasons for this behavior is device held in reset by some external circuit. Like reset supervisor, or just pull-down on accident.

ataradov, 2019-03-20:
If you can provide the schematic. And double check that everything is powered and connected correctly (connectors are not mirrored, for example).

NoeelMoeskops, 2019-03-21:
So, I tried reading an empty ATSAMD51J20A, and it works fine! Also flashing software works. The only problem is when the software is flashed upon the MCU, EDBG always fails to interact with it. `0x00 (count = 0, value = 0` on reading and `0x0d (count = 1, value = 4)` when trying to read the fuses. It all works fine on an empty chip tho. But once it is flashed it seems to become uncooperative. 

My colleges are able to interface with it regarding of it being flashed with atmel-studio.

ataradov, 2019-03-21:
Can you share the binary that causes this behavior? Previously device that goes to sleep immediately would have problems like that, but this should be addressed in this version.

NoeelMoeskops, 2019-03-22:
Thank for you help @ataradov, but I'm unable to provide the binary as this it is a commercial product. You assumption of the processor going to sleep after boot is right.

NoeelMoeskops, 2019-03-22:
I flashed different software on the MCU using MPLAB, and I can now successfully read and write it using EDBG. But once the software with the immediate sleep in flashed, it becomes  uncooperative.

ataradov, 2019-03-22:
Do you have a reset line connected to the debugger?  What sleep mode are you using?

I will try again, but this release should not care what is the sleep mode, since it uses reset extension to intercept normal application startup. But that will fail if the reset line is not connected.

ataradov, 2019-03-22:
I tried with STANDBY, HIBERNATE, BACKUP modes and I can't reproduce this. 

I would look into this, If you could remove all the application logic from your binary and just do the sleep part. Otherwise I won't be able to help you, since there is no way for me to reproduce this.

NoeelMoeskops, 2019-03-25:
The reset line is connected to the MCU. I will have a chat with my colleagues and discuss if we can send you a minimal binary of the application that just involves the sleep function.

ataradov, 2019-03-26:
I had another idea. It is possible that your reset line has high capacitance, and the reset pulse is too short and does not actually reset the MCU. I extended it by 10 ms. 

Try with the latest build and see if this helps.

NoeelMoeskops, 2019-03-26:
Tried build 57 (gcc and clang) but the results where the same. Going to recreate the problem using only the sleep function and will send you the files.

NoeelMoeskops, 2019-03-26:
Build the application with only using hibernation and there doesn't seem to be no a problem interfacing it afterwards with build 57 or 56. So I think its something else, not sure what tho.

thijsconnecterra, 2019-03-27:
Thanks for all the help ataradov. I'm a colleague of Noeel, and I narrowed the problem down to the root cause. It seems like edbg is unable to erase the chip or program it if the security bit is set. The security bit is supposed to prevent the chip from being read, but not from being erased/programmed.

The error I'm getting when I'm trying to program is also slightly different:
`Error: invalid response while writing the register 0x0d (count = 1, value = 4)`

As some additional information:
In Atmel Studio, I am able to erase the chip, but not "start without debugging" with the security bit firmware, if this helps you at all.

NoeelMoeskops, 2019-03-27:
`NVMCTRL->CTRLB.reg = NVMCTRL_CTRLB_CMD_SSB | NVMCTRL_CTRLB_CMDEX_KEY;` to set the security bit

ataradov, 2019-03-27:
Yes, a simple sequence "edbg -b -t atmel_cm4v2 -k" and "edbg -b -t atmel_cm4v2 -e" fails. 

I will investigate and fix that.

ataradov, 2019-03-28:
I just pushed some changes. This should fix your issue.

NoeelMoeskops, 2019-03-28:
yep works, thanks for your help!

closed 2019-03-28 by NoeelMoeskops

blietaer, 2019-12-31:
Same issue here (SAM E70 on MCLV2 with ATMEL-ICE:

`edbg -b -t same70 -epv -k -f bin/MCLV-2.X.production.elf                                                                                                                                             
Debugger: ATMEL Atmel-ICE CMSIS-DAP J42700005341 1.0 (SJ)
Clock frequency: 16.0 MHz
Target: SAM E70Q21 (Rev B)
Erasing...  done.
Programming.........,................................................................................................ done.
Verification................................................................................................... done.
Locking...  done.`
it seem to show a perfect programming, but then indeed the MCU seems uncooperative afterwards.
But in my case, MPLAB can't recover it here...
I wish I could do a kind of erase....

Still same message then:
`edbg -b -t same70 -epv -k -f bin/MCLV-2.X.production.elf                                                                                                                                             
Debugger: ATMEL Atmel-ICE CMSIS-DAP J42700005341 1.0 (SJ)
Clock frequency: 16.0 MHz
Error: invalid response while writing the register 0x00 (count = 0, value = 7)`

Any idea what I am doing wring?

ataradov, 2019-12-31:
1. You are programming an ELF file. EDBG only supports raw binary files.
2. Once device is locked, the only way to unlock it is though assertion of the  ERASE pin. Unlock through the SWD is impossible, since locking disables the SWD interface.

ataradov, 2019-12-31:
Basically don't lock it if you are just debugging things.

blietaer, 2020-01-02:
1. Wow. My oversight indeed.
2. Yup, ERASE pin solves it all.
Thank you.
Still I wish I could use this nice EDBG CLI tool in my setup so I could call it from a one-line in Makefile... :(

ataradov, 2020-01-02:
Why do you need to lock the part in a first place? Keep it unlocked during development and you won't have to use ERASE pin.