ataradov/edbg #80: ATSAMD21G16L does not run after a 'successful' program

Lonney, 2019-09-09:
After manually adding its entry, programming an `ATSAMD21G16L` with an Atmel-iCE and a hex file (~34kb) _appears_ to work fine, including erasing and verifying.

However, the code on the chip does not run afterwards.

If I program the same device, with an Atmel-iCE, using Atmel Studio 7, with the same hex file, it runs just fine.

Dumping the memory from the device, after either program, reveals that the data is identical up until offset `0x00008420` (around where I expect the end of the hex file data to be).

After that though, the portion of the empty memory is different:

**Dump after EDBG programming**
[image]

**Dump after Atmel Studio 7 programming**
[image]

Is there some other option/configuration step I'm missing?

ataradov, 2019-09-09:
Is this dump of the raw data? This looks like a HEX file, and there is no point in looking at it in a hexadecimal view, it is just a text file. But you need to specify a binary file for the EDBG.

Make sure you specify a binary file as an argument.

Lonney, 2019-09-10:
Yes, these are reads of the chip flash using `--target samd21 --read dump.hex`.

> Make sure you specify a binary file as an argument.

Do you mean for programming or reading or both?

Reading it with `--target samd21 --read dump.bin` instead gives the same result at the end of the file:
[image]

Compared with that of the flash from a working chip:
[image]

ataradov, 2019-09-10:
If reading gives this result, it means you have programmed a HEX file in a first place. And it got interpreted as a binary. 

You need to specify raw binary file for both reading and writing.

Lonney, 2019-09-10:
Ah, apologies: for some reason I thought I remembered reading that EDBG supported programming with .hex files!

Using a .bin instead works as expected for this device, thank you!

closed 2019-09-10 by Lonney