ataradov/edbg #85: DAP_INFO_CAPABILITIES should be available

UweBonnes, 2019-11-27:
Hello,

I am trying to get the Blackmagic Debugger running with CMSIS_Dap adapters. I found your edbg a good start. As the DAP_PORT_JTAG may bot be implemented, is is needed outside of dap.c to know of "J" is in the return of ID_DAP_INFO/DAP_INFO_CAPABILITIES.

What would you propose to export these values?

ataradov, 2019-11-27:
Yes, I think the fix for zero-length strings issue will do that automatically. I plan to add the ability to request any information and do the debugger info formatting from edbg.c. This will expose all those things.

ataradov, 2019-11-28:
This is addressed in the most recent commit. There is now a dap_info() call that will return necessary information and capabilities definitions are also moved into the header file.

closed 2019-11-28 by ataradov