ataradov/edbg #88: Has anyone packaged edbg as a homebrew formula?

travisgriggs, 2019-12-31:
More of a question than an issue... but didn't know where else to ask.

ataradov, 2019-12-31:
I'm not aware of anything like that. And I don't have any Apple hardware, so I don't even know if MAC port is still functional.

I'm not sure how many people will see this question.

ooxi, 2020-01-01:
I'm not familiar with the Apple ecosystem. Can packaging for brew be automated? And is there a difference between brew and homebrew?

ataradov, 2020-01-01:
'Homebrew' is the build system, 'brew' is the command it uses. But that's as much as I know.

I assume homebrew uses some sort of description/manifest file that tells it how to build the application. Again, should not be too hard, but without the access to the OS for experiments it might be though.

travisgriggs, 2020-01-06:
See https://discourse.brew.sh/t/is-this-a-good-first-formula-project/6571.

Is having a versioned tarball available something you guys want to provide? Otherwise, I could make a fork I guess and add some tags. Correct me if I'm wrong, it's just "latest master" and the version is kept inside of edbg.c?

ataradov, 2020-01-06:
I don't want to manage tags or tarballs. I don't even increment the version number in the code consistently. New minor variations in the targets are just added without version increment.

We could potentially add creation of a tarball to the Travis CI. It already prepares pre-build binaries. There is no reason it could not package some files. But I guess that would involve telling homebrew where to get those new tarballs. In this case Travis build number could act as a version identifier.

Having a fork is also fine, but that would be outdated pretty fast.

ooxi, 2020-01-06:
Does the tar have to be immutable? Otherwise you could use [master.tar.gz](https://github.com/ataradov/edbg/archive/master.tar.gz) which always points to the newest commit in master.

ataradov, 2020-01-06:
Oh, I did not know that GitHub has this link. 

It would be better to use that, but I'm afraid homebrew will want to have a version number somewhere in its manifest files. One workaround would be to have the same fixed version number and just force and update. EDBG builds real fast, so it should not be a problem.

FRVisser, 2020-10-20:
As stated by [Homebrew documentation](https://docs.brew.sh/Acceptable-Formulae#stable-versions):
> Formulae in the core repository must have a stable version tagged by the upstream project. Tarballs are preferred to Git checkouts, and tarballs should include the version in the filename whenever possible.

Looking at the releases in [bintray](https://dl.bintray.com/ataradov/edbg/master/) include some versioning. Are these generated by Travis CI? Incrementing every run?

ataradov, 2020-10-20:
All the files are generated by the https://github.com/ataradov/edbg/blob/master/.travis.yml script.

The incrementing number in the directory name comes from the TravisCI build number. They should be monotonically incrementing for each build. 

The numbers at the end of the file name are a partial commit hash.

closed 2020-11-21 by ataradov