ataradov/edbg #89: Does not build out of the box on Catalina

travisgriggs, 2019-12-31:
I have a new Catalina install, XCode and developer tools installed. But I get the following

```
ENG-1510:edbg travisg$ make all
git clone git://github.com/signal11/hidapi.git
Cloning into 'hidapi'...
remote: Enumerating objects: 2006, done.
remote: Total 2006 (delta 0), reused 0 (delta 0), pack-reused 2006
Receiving objects: 100% (2006/2006), 2.73 MiB | 2.37 MiB/s, done.
Resolving deltas: 100% (1172/1172), done.
cd hidapi && ./bootstrap
+ autoreconf --install --verbose --force
./bootstrap: line 2: autoreconf: command not found
make: *** [hidapi/mac/.libs/libhidapi.a] Error 127
```

ataradov, 2019-12-31:
I did not do that hidapi-based port and I'm not too happy about its use. I personally would just use OS APIs directly. But I didn't do the port and I don't have Apple hardware to do a proper port.

You need to figure out what autoreconf is and where to get it. It is probably something from autoconf BS.

travisgriggs, 2019-12-31:
Here's a simplified approach to doing this.

brew install hidapi

Change Makefile as such:

```
ENG-1510:edbg travisg$ git diff
diff --git a/Makefile b/Makefile
index ae7035f..ee94c6b 100644
--- a/Makefile
+++ b/Makefile
@@ -28,11 +28,11 @@ else
   ifeq ($(UNAME), Darwin)
     BIN = edbg
     SRCS += dbg_mac.c
-    LIBS += hidapi/mac/.libs/libhidapi.a
+    LIBS += /usr/local/lib/libhidapi.a
     LIBS += -framework IOKit
     LIBS += -framework CoreFoundation
-    HIDAPI = hidapi/mac/.libs/libhidapi.a
-    CFLAGS += -Ihidapi/hidapi
+    HIDAPI = /usr/local/lib/libhidapi.a
+    CFLAGS += -I/usr/local/include/hidapi
   else
     BIN = edbg.exe
     SRCS += dbg_win.c
```

ataradov, 2019-12-31:
In general requiring third-party libraries to be obtained using system tools is my preferred approach.

But I won't integrate this right away, since I'm pretty sure that this will break automated TravisCI build. It would have to be changed to install brew somehow. I'm not sure it has it by default. And I don't have time to investigate this right now.

ataradov, 2019-12-31:
It looks like TravisCI does have homebrew installed by default. So it should not be that hard to switch it over. But I'm a bit busy now to mess with this.

ooxi, 2020-01-01:
I can change the Travis CI build to use brew, however I also don't have Apple hardware the verify the resulting binary

@travisgriggs would you be available to check if the result still works?

ataradov, 2020-01-01:
HIDAPI is now build using brew.

closed 2020-01-01 by ataradov