ataradov/edbg #92: microbit support

markfirmware, 2020-03-16:
Is it feasible for me to add support for the microbit? It uses a cm0 nrf51822. I looked at the cm0p support and I’m not sure what would need to be changed. Thanks, Mark.

ataradov, 2020-03-16:
You would need to change basically everything. Target code is very target specific. 

So to add a new target you need to take target_atmel_cm0p.c as a template, copy it, remove all the code from all the functions and re-implement it for that specific target.

But the first thin to check that micro-bit programmer chip is actually CMSIS-DAP compliant. Check that it shows up if you do 'edbg -l'.

markfirmware, 2020-03-16:
Attached debuggers:
      9900000039494e45006f00050000003e0000000097969901 - ARM DAPLink CMSIS-DAP
      ATML2637010000000000 - Atmel Corp. EDBG CMSIS-DAP

The first is a microbit and the second is an atsame70-xpld.

I’ll look more closely at target_atmel_cm0p.c.

Thanks!

ataradov, 2020-03-16:
Yes, it looks like the debugger part is fine. You can try to run something like 'edbg -b -t samd21 -e'. This is a command to erase the device, but it will not do anything for Nordic device, of course. But it will at least show that edbg can talk to the debugger.

It will likely complain that DSU_DID is not recognized. If it does this, then you are in business.

markfirmware, 2020-03-16:
Debugger: ARM DAPLink CMSIS-DAP 9900000039494e45006f00050000003e0000000097969901 1.0 (S)
    Clock frequency: 16.0 MHz
    Error: unknown target device (DSU_DID = 0x00000000)

Ok, I will proceed. Thanks.

ataradov, 2020-03-16:
Yep, so the link with the target actually works. It tried to read the word at the location of the DSU.DID register, and the result was 0, since there is nothing at that place in Nordic.

Now you basically need to read through the code of the Cortex-M0+ target, understand what it does and write equivalent code for your chip.

markfirmware, 2020-03-16:
Ok, I’ll keep this issue open for any follow-up questions.

closed 2020-11-21 by ataradov

markfirmware, 2020-11-24:
Yeah, it is going to be quite a while before I get back to this. Thanks for your support so far!