ataradov/edbg #96: How to get target voltage?

biboc, 2020-07-30:
Hi,

I have a samr21/30 target board. I want to get the target voltage to check if my target is correctly powered before flashing?
How could I add an arg to read target voltage only?

Thanks,

ataradov, 2020-07-30:
There is no standard way to read the target voltage. Every debugger hardware implements it differently and many debuggers do not have a way to do that at all.

You can try to guess by reading the values of the RESET, SWDIO and SWCLK pins. It will not tell you the exact voltage, but at least may indicated that the target is there. But again, if specific debugger has pull-ups on those pins from the local supply, then it will not work.

biboc, 2020-07-31:
I have an EDBG-based boards and a Atmel-ICE debugger

Yes, that's true, it is what I need, know if my target is correctly powered. 
I would like to use a command to read the value instead of an ADC or another tool
I try to find a link between the documentation (http://ww1.microchip.com/downloads/en/DeviceDoc/50002630A.pdf) and edbg program you wrote but I can't. 
So which command should I send to edbg to read value of a pin?

ataradov, 2020-07-31:
Part of the EDBG that deals with anything non- CMSIS-DAP is proprietary and not documented. So I have no idea what command is responsible for that.

ataradov, 2020-07-31:
I don't think this document is complete. It looks like it is missing a bunch of definitions, unless I'm missing something. 

The only reference to the target voltage I can see is in the section 5.5.5 "Housekeeping SET/GET Parameters". But it refers to the command  VTG, which is not defined. But logically it would be either 4, if commands are just sequential, or 0 if each context has its own command. Or something entirely different - you will have to experiment.

The way you sent those commands I think is somehow over the command 4.1.3 "GET". And the way this is set is through the vendor-specific CMSIS-DAP command described in 3.2 "CMSIS-DAP Vendor Commands"

How exactly it all works - I don't know, it is a mess and you would have to just experiment.

closed 2020-11-21 by ataradov