ataradov/edbg #97: Programming Speed Discussion

limpkin, 2020-08-01:
Hello and first of all thanks a lot for the great work!

I'm currently evaluating programming options for an atsamd21 and am benchmarking programming speeds for different tools and different programmers. At the moment, I'm noticing a massive difference between Atmel Studio and tools on linux (see below).
I would be therefore curious to know if you have done in the past some similar benchmarks?

My current results:

Atmel Studio for different SWD clocks, 142652 bytes firmware, program and verify:
Atmel-ICE:
- 8MHz (not recommended): 12s
- 4MHz: 13s17
- 2MHz: 14s49
- 1MHz: 17s40
- 500kHz: 22s31
JLink:
- 8MHz (not recommended): 7s5
- 4MHz: 8s
- 2MHz: 8s61
- 1MHz: 9s93
- 500kHz: 13s

Edbg on Ubuntu on Virtualbox, same firmware, erase, program and verify:
command line: sudo ./edbg-b77-LINUX_GCC-04b7508 -b -epv -t samd21 -f mini_ble.bin -c 8000
Atmel-ICE:
- 16MHz (not recommended): 1m01
- 8MHz (not recommended): 1m01
- 4MHz: 1m04
- 2MHz: 1m04

JLink: doesn't have CMSIS-DAP

OpenOCD on Ubuntu on Virtualbox, same firmware, erase, program & verify:
command line for jlink: sudo openocd -f samd21_jlink.cfg  -c "init; targets; program mini_ble.bin verify reset; shutdown"
command line atmel ice: sudo openocd -f samd21_ice.cfg  -c "init; targets; program mini_ble.bin verify reset; shutdown"

Atmel-ICE:
- 2MHz:  1m04
- 1MHz: 1m00
- 400kHz: 1m02

JLink:
- 8MHz: FAIL
- 4MHz: FAIL
- 2MHz: FAIL
- 1MHz: 1m27
- 500kHz: 1m27

samd21_ice.cfg file contents:
source [find interface/cmsis-dap.cfg]
set CHIPNAME at91samd21g18
set ENDIAN little
set telnet_port 0
source [find target/at91samdXX.cfg]
adapter_khz 2000

samd21_jlink.cfg file contents:
interface jlink
transport select swd
set CHIPNAME at91samd21e18
set ENDIAN little
set telnet_port 0
source [find target/at91samdXX.cfg]
adapter_khz 2000

ooxi, 2020-08-01:
You might be interested in [expected flashing speed](https://github.com/ataradov/free-dap/issues/7) in the free-dap repository. For mee using edbg + free-dap is about 2x slower than Atmel Studio + Atmel-ICE.

I really recommend you trying out free-dap instead of OpenOCD, it's much faster and easier to use (for flashing / verification).

limpkin, 2020-08-01:
oh i'd be interested to benchmark with that programmer as well...

ataradov, 2020-08-01:
I will need to have a closer look at your numbers, but something is strange here. Here is a test on Linux with SAMD21 Xplained Pro (onboard EDBG):

```
$ time edbg -b -epv -t samd21 -f test.bin -c 8000
Debugger: ATMEL EDBG CMSIS-DAP ATML2130021800008472 03.19.01A3 (S)
Clock frequency: 8.0 MHz
Target: SAM D21J18A (Rev C)
Erasing...  done.
Programming................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................. done.
Verification................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................. done.

real	0m8.895s
user	0m0.100s
sys	0m0.872s
```

ataradov, 2020-08-01:
That programmer from Tindie would be slower, since it is only USB FS. So it is limited to 64 Kbytes/s transfer speed, and 64-byte frames. It also can't do 8 MHz. It will top out at around 6 MHz or so.

But I will try it to later after I get Atmel-ICE and see if there are any issue with it.

Once of the issues with various debuggers is that they sometimes accept clock command while interface is connected and sometimes not.

ataradov, 2020-08-01:
Also, for completeness, the expected time it takes to perform flash operations here is: 240 ms (chip erase) + 1.2 ms * 558 (page erase) + 2.5 ms * 2229 (page programming) = 6.48 seconds. This is the minimal number and everything else is protocol overhead.

limpkin, 2020-08-01:
@ataradov how big is your test.bin? I'm testing with a particularly large fw file...
Somehow, using JLinkExe I managed to get 4 (!) seconds programming time....

ataradov, 2020-08-01:
It is exactly as your mentioned - 142652 bytes.

The times are a mix of maximum and typical values, so the actual time on typical silicon may be shorter, but there may be outliers. The specifications for the maximum values are notoriously big, since Flash manufacturing is a pain and highly unpredictable. So 4 seconds is not out of the question for some (most?) parts.

Also, I don't think you do chip erase in case of Jlink (command line version). You don't don't need "-e" in case of edbg either. Programming operation will erase the affected area anyway. I tested and it does take about 200 ms.

limpkin, 2020-08-01:
wow that's odd indeed... could it come from the fact that I'm running this on a VM?

ataradov, 2020-08-01:
Ok, some strange results.  I took edbg-b77-LINUX_GCC-04b7508 for consistency. I will be testing with SAM D11, since SAM D11 Xplained Pro has on-board EDBG and a connector fro the external programmer.  It is only 16K device. I'm using a full 16K image. The command is '-b -t samd11 -epv -f test_16k.bin -c 8000' for all tests.

On-board EDBG:  0m4.699s
Atmel-ICE: 0m1.175s
Free-DAP (Exact board from Tindie): 0m4.877s

I think I remember there were some differences between how on-board EDBG and Atmel-ICE EDBG accept the clock. Let me dig into this. There should be no difference between them.

ataradov, 2020-08-01:
VM may also be a factor, I'm not sure.

ataradov, 2020-08-01:
Yes, on-board EDBG for whatever reason wants the clock to be configured only when the link is already open. I pushed a small patch. Wait until a new version builds and try that.

It is possible that it just depends on the version of the firmware, since the firmware inside Atmel-ICE EDBG and on-board EDBG is basically the same.

limpkin, 2020-08-01:
actually now that I think about it I don't think it'd come from the VM as the 4s were gotten on that very VM with JLinkExe... I'll try the newer version then :)

ataradov, 2020-08-01:
The new version is available. 

Also, record the version of the firmware in your Atmel-ICE. My is "Debugger: ATMEL Atmel-ICE CMSIS-DAP J41800003813 1.0 (SJ)"

ataradov, 2020-08-01:
But if you used it with Atmel Studio, it is probably updated to the latest version anyway.

limpkin, 2020-08-01:
No big success i'm afraid...

time sudo ./edbg-b78-LINUX_GCC-dca761e  -b -epv -t samd21 -f mini_ble.bin -c 8000
Debugger: ATMEL Atmel-ICE CMSIS-DAP J42700004902 1.0 (SJ)
Clock frequency: 8.0 MHz
Target: SAM D21G18A (Rev D)
Erasing...  done.
Programming................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................. done.
Verification................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................. done.

real	1m1,370s
user	0m0,129s
sys	0m0,434s

ataradov, 2020-08-01:
Time each individual process (erase, program, verify). What is  the time breakdown?

I would actually looks with the scope at the actual SWD clock. 

It may be after all the VM. May be some differences in the used protocols or something make it different for sifferent debuggers.

ataradov, 2020-08-01:
Also, look in the dmesg output after Atmel ICE has attached. Does it actually enumerate as a HS device?

limpkin, 2020-08-01:
It's enumerated as full-speed... 
Due to virtualbox innerworkings, i'll setup a new VM and get back to you... thanks!
If that's the root cause, which (cheap) programmer handling high-speed do you recommend for its highest flashing speed? :)

ataradov, 2020-08-01:
Well, Atmel-ICE is quite good as far as performance goes. When it is enumerated as HS device, of course.

The only other HS capable CMSIS-DAP programmer I know is LPC-Link 2. It comes by default with NXP proprietary firmware, but there is a firmware from NXP that turns it into a CMSIS-DAP debugger.

But I have not looked at the market for debuggers for a while.

limpkin, 2020-08-02:
thanks for the feedback!

we're definitely better: 

time sudo ./edbg-b78-LINUX_GCC-dca761e -b -epv -t samd21 -f mini_ble.bin -c 8000
Debugger: ATMEL Atmel-ICE CMSIS-DAP J42700004902 1.0 (SJ)
Clock frequency: 8.0 MHz
Target: SAM D21G18A (Rev D)
Erasing...  done.
Programming................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................. done.
Verification................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................. done.

real	0m22,943s
user	0m0,126s
sys	0m0,366s

ataradov, 2020-08-02:
There is still probably some VM slowness. I can't experiment with D21 and Atmel-ICE, but EDBG on Xpro should give the same exact speed and on a real PC is about two times faster.

ooxi, 2020-08-02:
@limpkin why are you using a VM at all? edbg is available for Linux, Mac and Windows. Might be an idea to use edbg directly on host just to rule out any VM related issues.

If edbg does not run on your host, drop me a message and we'll try to figure something out.

limpkin, 2020-08-02:
actually for me the next step will be to run edbg on the final mass programming setup and then report here the programming speeds. this is what will drive the programming tool & program decision :)

biboc, 2020-08-27:
USB Hub can affect your programming speed, try to plug atmel ICE in your laptop

ataradov, 2020-08-27:
There is no way USB hub can slow things down enough. Majority of time during programming is actually spent waiting on the flash. And that is done though polling of flags. It does not matter if you don't poll as frequently. The data transfer will still be fast.

biboc, 2020-08-27:
My test showed that my usb Hub slowed speed down, it was detected as full speed instead of high speed
If USB hub is High speed, I agree that this should not affect programming speed

ataradov, 2020-08-27:
Well, yes, if you have a FS hub, then it will turn all devices in FS. That's a very different story, and such a hub belongs in a trash bin.

closed 2020-11-21 by ataradov