ataradov/edbg #101: Migrate to GitHub Actions?

ooxi, 2020-11-07:
Unfortunately [travis-ci.org will shut down in December](https://mailchi.mp/3d439eeb1098/travis-ciorg-is-moving-to-travis-cicom?e=%5BUNIQID%5D). While Travis CI encourages moving to travis-ci.com, I do not recommend this move, since the [open source build minutes are severely limited](https://blog.travis-ci.com/2020-11-02-travis-ci-new-billing).

I therefore propose to migrate the CI to GitHub Actions. There are other alternatives like Circle CI, however I am most familiar with GitHub Actions.

Would moving to GitHub Actions be ok for you? I would propose a pull request.

ataradov, 2020-11-07:
I already moved to travis-ci.com, but I don't really care what builds those binaries. GitHub Actions would work for me.

If there is a way to publish a GuHub release, it would be fine too instead of using Bintray.

ataradov, 2020-11-21:
BTW, MAC OS build now fails with the following error:
```
3.06s$ make all
gcc -I/usr/local/include/hidapi -W -Wall -Wextra -O2 -std=gnu11 dap.c edbg.c target.c target_atmel_cm0p.c target_atmel_cm3.c target_atmel_cm4.c target_atmel_cm7.c target_atmel_cm4v2.c target_mchp_cm23.c target_st_stm32g0.c target_gd_gd32f4xx.c  dbg_mac.c -framework IOKit -framework Foundation -framework CoreFoundation /usr/local/lib/libhidapi.a -o edbg
Undefined symbols for architecture x86_64:
  "_NSAppKitVersionNumber", referenced from:
      _hid_init in libhidapi.a(hid.o)
ld: symbol(s) not found for architecture x86_64
```
I'll disable it for now, since I have not idea about any MAC specific stuff. And I don't care that much about apple to waste a lot of time investigating this.

ooxi, 2020-11-22:
@ataradov unfortunately I'm not that familiar with mac os myself, but I'll try to fix it

ataradov, 2020-11-22:
Don't waste a lot of time on that. MAC port is not great to begin with. It relied on hidapi library and that may have been broken. And I personally don't care to waste any resources on apple.

ooxi, 2020-12-09:
Btw. the GitHub Actions port is almost done. However GitHub Releases does not work properly yet but I have hopes of fixing it

ooxi, 2021-01-11:
@ataradov GitHub Releases is more complex than initially estimated :-( the reason for this is that you have to separately create a release and then attach binaries. Which is quite a hassle.

Would you be OK with staying with bintray for now and maybe migrate to GitHub Releases later?

ataradov, 2021-01-11:
Sure, I'm fine with whatever.

ataradov, 2021-02-04:
Bintray is going away May 1st. https://jfrog.com/blog/into-the-sunset-bintray-jcenter-gocenter-and-chartcenter/

ooxi, 2021-02-07:
Aw man :-( I'm sure I can figure GitHub releases out :-)

ataradov, 2021-02-07:
I'm considering another option. Just make a simple Python script that would just check the repo periodically. And run that on a dedicated server. Then serve the binaries from the same server. I have a server that can do this. It does not have a ton of bandwidth, but can probably support downloads for this.

ataradov, 2021-02-07:
This would eliminate MAC builds, but MAC is such a PITA that I don't really care.

ooxi, 2021-02-12:
That would be great for providing releases! I would still like to have CI for pull requests, but serving releases from your website would be even better, since you have full control and it's thus from a verified origin

ataradov, 2021-03-29:
CI is moved to GitHub actions. Binaries would be published automatically later, once I figure out the best way to do that.

closed 2021-03-29 by ataradov