ataradov/edbg #106: Big Sur compilation issues

travisgriggs, 2021-03-19:
For a fresh compile on MacOS Big Sur, I found that I needed to add

ataradov, 2021-03-19:
I pushed the change, let's see if it builds on TravisCI server too.

ataradov, 2021-03-19:
Travis CI build still fails. Brew is a problem now. It for inexplicable reason tries to build java, which takes longer than 10 minutes and the build gets killed.

I kept the change in a Makefile, but the automated build were disabled again, until hidapi dependency is either removed, or replaced with some better pre-built source.

Here are the last messages in a log:
```
==> java has been moved to Homebrew.
To uninstall the cask, run:
  brew uninstall --cask --force java
==> Installing java...
Warning: You are using macOS 10.13.
We (and Apple) do not provide support for this old version.
You will encounter build failures with some formulae.
Please create pull requests instead of asking for help on Homebrew's GitHub,
Twitter or any other official channels. You are responsible for resolving
any issues you experience while you are running this
old version.
==> Downloading https://download.java.net/java/GA/jdk14.0.2/205943a0976c4ed48cb16f1043c5c647/12/GPL/openjdk-14.0.2_osx-x64_bin.tar.gz
==> Downloading https://hg.openjdk.java.net/jdk-updates/jdk15u/archive/jdk-15.0.2-ga.tar.bz2
Warning: Your Xcode (9.4.1) is outdated.
Please update to Xcode 10.1 (or delete it).
Xcode can be updated from the App Store.
Warning: A newer Command Line Tools release is available.
Update them from Software Update in the App Store or run:
  softwareupdate --all --install --force
If that doesn't show you any updates, run:
  sudo rm -rf /Library/Developer/CommandLineTools
  sudo xcode-select --install
Alternatively, manually download them from:
  https://developer.apple.com/download/more/.
==> ./configure --without-version-pre --without-version-opt --with-version-build=7 --with-toolchain-path=/usr/bin --with-boot-jdk=/private/tmp/openjdk-20210319-3108-1hyl0q6/jdk15u-jdk-15.0.2-ga/boot-jdk/Contents/Home --with-boot-jdk-jvmargs=-Duser.home=/Users/travis/Library/Caches/Homebrew/java_cache --with-debug-level=release --with-native-debug-symbols=none --with-jvm-variants=server --with-sysroot=/Library/Developer/CommandLineTools/SDKs/MacOSX10.13.sdk --with-extra-ldflags=-headerpad_max_install_names --enable-dtrace
==> make images
No output has been received in the last 10m0s, this potentially indicates a stalled build or something wrong with the build itself.
Check the details on how to adjust your build configuration on: https://docs.travis-ci.com/user/common-build-problems/#Build-times-out-because-no-output-was-received
The build has been terminated
```