ataradov/edbg #107: Cannot Program SAMD10 Xplained Mini

DaAwesomeP, 2021-04-28:
Hello,

Board was working fine for some time, but now that my program has grown in size, I am having trouble programming it. It uploads the larger program just fine but does not work after that. It responds in Microchip/Atmel Studio just fine, and if I erase the board from there then I can program it with this tool. This is using the onboard programmer; there is only a USB cable involved.

OpenOCD seems to be affected by the same issue (I know this is a different project but could be relevant if they work in similar ways).

```
Debugger: Atmel mEDBG CMSIS-DAP ATML2378032700000781  (SJ)
Clock frequency: 16.0 MHz
Error: invalid response during transfer (count = 0/1, status = 7)
```

ataradov, 2021-04-28:
This looks like it fails to even identify the part.

Do you disable reset pin function in your application? Does your application use sleep? Can you attach the file that gets it into that state?

ataradov, 2021-04-28:
I'm not sure what studio would do differently than what this tool is doing. The only thing I can think of is that EDBG chip has a POWER_ENABLE output that controls the power to the MCU, which may be helpful in recovering from the reset pin reconfigured as a GPIO. But this is a proprietary extension to CMSIS-DAP protocol, and I have no idea how to control that pin.

DaAwesomeP, 2021-04-28:
Thanks for your fast response!

OK, so I was a dummy and was using PA28 (reset) as GPIO (I completely understand why not to do that, just missed it while assigning pins). However, not using that pin did not fix the issue. Reading through the datasheet more thoroughly about which pins the debugger actually uses, I see that I should also stay away from PA30 and PA31 (SWCLK and SWDIO). Though it also says "a debugger cold-plugging or hot-plugging detection will automatically switch the SWDIO port to the SWDIO function." I presume this is something in a proprietary CMSIS-DAP extension? The DSU section of the datasheet seems to describe in detail how to trigger a hot-plugging event, but I would imagine it is some special function of the mEDBG unless you can directly control SWDIO state changes.

The reset pin broken out on the board seems to work fine (both when using PA28 as GPIO and not), so without looking at the schematic I have a feeling that is going to the mEDBG chip to cut power to the SAMD, not the SAMD directly. 

I am able to replicate the issue with the base project I generated in Atmel START (without adding any of my own code). I have attached a zip here with binaries too. To rebuild, enter the `gcc` directory and run `make` (standard gcc arm toolchain dependencies apply). I did modify `main.c` to use the correct pins but that's it. This project was generated by adding the modules I planned to use, labeling and setting the pins I planned to use, and then downloading with GCC checked. Though I think it's probably just from making use of PA30 and/or PA31 so you probably have no need to test this.

I will remove my use of PA30 and PA31 and see if that solves the issue and report back, but that's probably it. Thank you for catching my careless error.

DaAwesomeP, 2021-04-28:
Yup, not using PA30 and PA31 solved the issue.

I'll close this since this isn't really a show-stopping issue as I reported it. It would be cool if it could do the hot-plugging, but it seems like that will be a proprietary extension to the mEDBG even though the high/low timings are shown in the datasheet (i.e. for if you made your own debugger from scratch).

ataradov, 2021-04-28:
No, the detection is implemented. If the reset pin is free to be a reset, then if SWCLK is low while reset is asserted and released, then the firmware execution would be interrupted. EDBG implements this. And given that at reset pins are configured as SWD, just having access to reset should be enough to recover from any situation.

I'll try that project.

DaAwesomeP, 2021-04-28:
Ah, we responded simultaneously!

I have quite a few of the boards on-hand, so I am happy to try things out if you do not have this particular board/debugger.

ataradov, 2021-04-28:
This was a logical error introduced in the last big refactoring of the DAP stuff. It did not matter on the boards that had SWD interface accessible without the reset extension procedure, but failed on the boards where SWD interface was disabled. This is now fixed. See if that works for you.

ataradov, 2021-04-28:
There is actually a much better solution, but it would need more testing as it would potentially affect multiple targets. I'll need to think about it for a bit.

ataradov, 2021-04-28:
Ok, it actually was not that scary. I've pushed a better fix.

DaAwesomeP, 2021-04-28:
OK, I downloaded the build from https://github.com/ataradov/edbg/commit/7f0e40934384450f36bb24d7e83e59b5f2a1e77f and it seems to work perfectly with the project I attached above! I double checked that it still doesn't work in OpenOCD, so your changes definitely fixed it! Thank you so much!

closed 2021-04-28 by DaAwesomeP