ataradov/edbg #109: SAML10/L11 Erasure programming failed

neomissing, 2021-05-06:
Test programming today and find it doesn't work? Is there a programming problem？

ataradov, 2021-05-06:
You would need to provide more details. What exactly does not work? What errors you are getting? How are you running the program? What debugger are you using? What target hardware? Howe are you connecting the debugger to the target?

neomissing, 2021-05-07:
I use IBDAP, Use the following command:edbg -b -t saml11 -pv -f Demo.bin,Has been in Programming......，Won't end.

ataradov, 2021-05-07:
The part before ',' is erase, which apparently works. The part after is supposed to be programming, which never completes.  I have never seen this happen before. One thing you can try is decrease the clock frequency.

I think I have IBDAP, so I will try that tomorrow.

ataradov, 2021-05-07:
Ok, there is no ','. Please quote the exact output when reporting bugs.

I can reproduce the issue. It does not depend on the debugger. I see the same behaviour with other debuggers. I'll try to figure out what is wrong.

ataradov, 2021-05-07:
The issue was related to L11 and its security stuff. It should be fixed now.

closed 2021-05-07 by ataradov