ataradov/edbg #111: JTAG primer

ooxi, 2021-05-16:
Thank you very much for investing the effort to make both edbg and free-dap JTAP capable! I would love to check out the implementation, however I'm not that familiar with actually using JTAG.

Before digging deeper into the guts of JTAG, what operations should work right now with edbg + free-dap and what target can be used for testing? I'm mostly working with SAM1x/2x chips and those don't seem to support JTAG, just SWD.

ataradov, 2021-05-16:
For JTAG there are two distinct modes. First one is the normal transfers. This is ARM specific, but more  standard, of course.

For this mode to work, simply replace
```
  dap_connect(DAP_INTERFACE_SWD);
```
with
```
  dap_jtag_configure(2, (int[]){ 4, 5 });
  dap_jtag_set_index(0);
```
Here 4 and 5 are IR lengths of the devices in the JTAG chain. Many ARM device (STM32f4xx in this case) have 2 devices in the chain. One is for boundary scan (5) and one for the ARM core (4). devices are declared in the order starting from the end of the chain (first device attached to TDI). And index is the index of the device corresponding to the ARM core (0 in this case, as the core is last in the chain).

After that the resto of the code would work as is. In my tests I used "dap_read_word(0xe0042000);". This is where ID of the device is located. And this worked for both JTAG and SWD mode only with the change above. I did not implement the whole driver, as I have no particular need for that device. I just used it as a test target.

If there are any other external devices in the chain, they need to be added too. If there is an actual need to support some JTAG-only device, then this would be passed through the command line.

The second mode is just bit banging, which would work for any device. OpenOCD uses this mode exclusively for everything.

The example of this mode is in the function dap_jtag_scan_chain(). It does some basic operation of resetting the JTAG TAP, shifting out the DR until it reads 0, those enumerating all IDCODEs for all the devices. It then measures the total length of the IR registers in the chain (9 in this case).

This is as much of the automated identification as you can do. It is impossible to identify individual IR lengths of the devices. You just need to have a huge database of IDCODEs and corresponding lengths.

JTAG mode here was added for testing of the free-dap implementation. I just did not want to throw away the effort completely, so I packaged it into the minimal changes that made sense. 

It would probably be changes a bit once there is a real need to support JTAG-only device. It is likely to be some FPGA rather than MCU though.

ooxi, 2021-05-18:
Thank you very much for such a detailed response. I'll try to play around with it a bit and will come back if I need additional support.

closed 2021-05-18 by ooxi

ataradov, 2021-05-30:
I've added a first real JTAG-only target. It is Lattice LCMXO2 FPGA. 

I had a project where I needed to program the FPGA from the MCU anyway, so I just ported that code to edbg.