ataradov/edbg #112: "Error: invalid response received" in ATSAMD11C14A

Sreedharbot, 2021-05-31:
Hi,

Actually, This is my 1st time in working with ARM Cortex Chip.. Moreover, I want to make a Free_CMSIS_DAP on my own. 
You can see my documentation.. By clicking [Here](http://fabacademy.org/2020/labs/bangalore/students/sreedhar-ramaswamy/ExtraWorks/Atsamd11c14/) where I refer how to build Free_CMSIS_DAP on my own and my failures while building it.

I use:
* Programmer - [Particle Debugger](http://fab.academany.org/2020/labs/ulb/students/quentin-bolsee/images/project_d11c_serial/jtag.jpg) for uploading the ".bin file"
* System - mac IOS (Big Sur)

I also having doubt in using the pins.. so far i have seen many  are using jtag 10 pins to program in particle debugger..but when i check continuity between the pins and jtag connector, all are connect expect the RST line.. [image]  

I also made a video regarding this issue"Error: invalid response received". [Here is the link .](https://youtu.be/HRbkgo7SMH8)

Things I want to know is.. 
> why i got this error indication while uploading the .bin file?  I checked my connections all with Digital Multimeter.. all are connected properly.

 I don't have small JTag connector, hence i used jumpers to connect them. I tried  uploading the  code from Jtag connector and also from the female header pins  both indicated same error ? I also checked continuity of my board Free_CMSIS_DAP there is no interconnection.

Please help me out to rectify this issue..

Moreover, please could you share how the EDBG works with Micro-controller.. Is there any other way to debug.. this...

ataradov, 2021-05-31:
EDBG support on MAC is incomplete, since I don't have any Apple hardware.

It has the report size hard coded:
```
static int report_size = 512; // TODO: read actual report size
```
You probably need to change that value to 64 for your device, since it is likely to be a USB FS device.

Sreedharbot, 2021-05-31:
After Changing 
> static int report_size = 512; // TODO: read actual report size 
in the dbg_mac.c

the error still occurs...

Sreedharbot, 2021-05-31:
After Changing the 
> static int report_size = 64; // TODO: read actual report size
 in dbg_mac.c 
The error still occurs..

ataradov, 2021-05-31:
Then you need to print out the values of the hid_buffer before hid_write() and after hid_read(). First byte must match. But more importantly you will know on what command it fails. Do that and provide the output.

```
int dbg_dap_cmd(uint8_t *data, int resp_size, int req_size)
{
  uint8_t cmd = data[0];
  int res;

  verbose("---\n");
  for (int i = 0; i < 16; i++)
    verbose("0x%02x ", data[i]);
  verbose("\n");

  memset(hid_buffer, 0xff, report_size + 1);

  hid_buffer[0] = 0x00; // Report ID
  memcpy(&hid_buffer[1], data, req_size);

  res = write(debugger_fd, hid_buffer, report_size + 1);
  if (res < 0)
    perror_exit("debugger write()");

  res = read(debugger_fd, hid_buffer, report_size + 1);
  if (res < 0)
    perror_exit("debugger read()");

  check(res, "empty response received");

  for (int i = 0; i < 16; i++)
    verbose("0x%02x ", hid_buffer[i]);
  verbose("\n");

  check(hid_buffer[0] == cmd, "invalid response received");

  res--;
  memcpy(data, &hid_buffer[1], (resp_size < res) ? resp_size : res);

  return res;
}
```

Sreedharbot, 2021-05-31:
### In the file dbg_mac.c..  there is  "No hid_write()  and  hid_read()".. 

### In 2 ways I have applied the  following code..

1.  In this method, I copy pasted the following code in the end of the dbg_mac.c file. Like wise, saved the file  and run edbg.
[image]

##  I obtained same result.... of "Error invalid response received"

2.  In this method,  I copy pasted the following code in the end of the dbg_mac.c file. Like wise as mentioned above. This time I have saved the file..  and run "make clean " and "make all".. in the edbg directory. 
* After executing  make all.. I obtained a message like this... 

[image]

Moreover, I was not able to find hid_write() and hid_read()..

ataradov, 2021-05-31:
The code here is for Linux version. Leave the existing mac code as is, just copy the part that prints the data.

Sreedharbot, 2021-06-01:
Okay.. 
I have edited dbg_mac.c
```
int dbg_dap_cmd(uint8_t *data, int resp_size, int req_size)
{
  uint8_t cmd = data[0];
  int res;

   verbose("---\n");
  for (int i = 0; i < 16; i++)
    verbose("0x%02x ", data[i]);
  verbose("\n");

  memset(hid_buffer, 0xff, report_size + 1);

  hid_buffer[0] = 0x00; // Report ID
  memcpy(&hid_buffer[1], data, req_size);

  res = hid_write(handle, hid_buffer, report_size + 1);
  if (res < 0)
  {
    message("Error: %ls\n", hid_error(handle));
    perror_exit("debugger write()");
  }

  res = hid_read(handle, hid_buffer, report_size + 1);
  if (res < 0)
    perror_exit("debugger read()");

  check(res, "empty response received");

   for (int i = 0; i < 16; i++)
    verbose("0x%02x ", hid_buffer[i]);
    verbose("\n");

  check(hid_buffer[0] == cmd, "invalid response received");

  res--;
  memcpy(data, &hid_buffer[1], (resp_size < res) ? resp_size : res);

  return res;
}
```
I have just edited the file by adding the verbose.. in the top and both.. (<b>Please correct me, if I am going it wrong ?</b>)

then, I run "make clean " and "make all"..

This time as you said I run the edbg in the edbg directory..

the code, i have used is 
```
./edbg -bpv -e -t samd11 -f sam_ba_Generic_D11C14A_SAMD11C14A.bin
```
this time, I got a strange output.. like this??..
```
sreedhar@Sreedhars-MacBook-Pro edbg % ./edbg -bpv -e -t samd11 -f sam_ba_Generic_D11C14A_SAMD11C14A.bin
---
0x00 0x01 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 
0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 
---
0x00 0x02 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 
0xff 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 
---
0x01 0x00 0x00 0x90 0x01 0xef 0xe9 0xfe 0x7f 0x00 0x00 0x41 0x50 0xd1 0x05 0x01 
0xff 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 
---
0x03 0x90 0x01 0xef 0xe9 0xfe 0x7f 0x00 0x00 0x46 0x50 0xd1 0x05 0x01 0x00 0x00 
0xff 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 
Error: invalid response received
```
Can you tell me what this HEX code trying to say..? , how do you decode this ?..  and how to rectify this issue?....

ataradov, 2021-06-01:
This looks like you have not actually corrected 512 -> 64. Do this change, recompile and try again.

The first byte of the request and the response must match. But this could only happen if you send the exact number of bytes the device is expecting.

Sreedharbot, 2021-06-01:
just forgot to do the change....

Right Now,  I did the change from 512 to 64 and ran "make clean",  "make all"

After executing "./edbg -bpv -e -t samd11 -f sam_ba_Generic_D11C14A_SAMD11C14A.bin"  

I received 
```
---
0x00 0x01 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 
0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 
---
0x00 0x02 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 
0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 
---
0x00 0x03 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 
0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 
---
0x00 0x04 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 
0x00 0x05 0x31 0x2e 0x31 0x30 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 
---
0x00 0xf0 0x2e 0x31 0x30 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 
0x00 0x01 0x11 0x2e 0x31 0x30 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 
Debugger: ARM DAPLink CMSIS-DAP 1093000031389eb81139333437353533a5a5a5a597969906 1.10 (S)
Clock frequency: 16.0 MHz
---
0x03 0x20 0x7a 0xb0 0xe6 0xfe 0x7f 0x00 0x00 0x9f 0xea 0x0f 0x09 0x01 0x00 0x00 
0x03 0x00 0x11 0x2e 0x31 0x30 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 
---
0x02 0x01 0xf0 0x76 0xb0 0xe6 0xfe 0x7f 0x00 0x00 0xa0 0x03 0x11 0x09 0x01 0x00 
0x02 0x01 0x11 0x2e 0x31 0x30 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 
---
0x04 0x00 0x00 0x80 0x80 0x00 0x20 0x7a 0xb0 0xe6 0xfe 0x7f 0x00 0x00 0xba 0xea 
0x04 0x00 0x11 0x2e 0x31 0x30 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 
---
0x13 0x00 0x20 0x7a 0xb0 0xe6 0xfe 0x7f 0x00 0x00 0xc1 0xea 0x0f 0x09 0x01 0x00 
0x13 0x00 0x11 0x2e 0x31 0x30 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 
---
0x11 0x00 0x24 0xf4 0x00 0x20 0x7a 0xb0 0xe6 0xfe 0x7f 0x00 0x00 0xcc 0xea 0x0f 
0x11 0x00 0x11 0x2e 0x31 0x30 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 
---
0x01 0x00 0x01 0x20 0x7a 0xb0 0xe6 0xfe 0x7f 0x00 0x00 0xd8 0xea 0x0f 0x09 0x01 
0x01 0x00 0x11 0x2e 0x31 0x30 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 
---
0x10 0x00 0x83 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0xa0 
0x10 0x00 0x11 0x2e 0x31 0x30 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 
---
0x10 0x80 0x83 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0xa0 
0x10 0x80 0x11 0x2e 0x31 0x30 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 
---
0x12 0x88 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0x9e 0xe7 0xff 0xff 0xff 0xff 0xff 
0x12 0x00 0x11 0x2e 0x31 0x30 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 
---
0x05 0x00 0x01 0x02 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 
0x05 0x01 0x01 0x77 0x14 0xc1 0x0b 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 
---
0x05 0x00 0x03 0x00 0x16 0x00 0x00 0x00 0x08 0x00 0x00 0x00 0x00 0x04 0x00 0x0f 
0x05 0x03 0x01 0x77 0x14 0xc1 0x0b 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 
---
0x05 0x00 0x03 0x01 0x12 0x00 0x00 0xa3 0x05 0x18 0x21 0x00 0x41 0x0f 0x00 0x00 
0x05 0x03 0x01 0x06 0x01 0x03 0x10 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 
Target: SAM D11C14A (Rev B)
---
0x05 0x00 0x03 0x01 0x10 0x00 0x00 0xa3 0x05 0x02 0x21 0x00 0x41 0x0f 0x00 0x00 
0x05 0x03 0x01 0x00 0x00 0x12 0x10 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 
---
0x05 0x00 0x03 0x01 0x12 0x00 0x00 0xa3 0x05 0xf0 0xed 0x00 0xe0 0x0d 0x03 0x00 
0x05 0x03 0x01 0x00 0x00 0x12 0x10 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 
---
0x05 0x00 0x03 0x01 0x12 0x00 0x00 0xa3 0x05 0xfc 0xed 0x00 0xe0 0x0d 0x01 0x00 
0x05 0x03 0x01 0x00 0x00 0x12 0x10 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 
---
0x05 0x00 0x03 0x01 0x12 0x00 0x00 0xa3 0x05 0x0c 0xed 0x00 0xe0 0x0d 0x04 0x00 
0x05 0x03 0x01 0x00 0x00 0x12 0x10 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 
---
0x05 0x00 0x03 0x01 0x10 0x00 0x00 0xa3 0x05 0x01 0x21 0x00 0x41 0x0d 0x00 0x02 
0x05 0x03 0x01 0x00 0x00 0x12 0x10 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 
Erasing...---
0x05 0x00 0x03 0x01 0x10 0x00 0x00 0xa3 0x05 0x01 0x21 0x00 0x41 0x0d 0x00 0x01 
0x05 0x03 0x01 0x00 0x00 0x12 0x10 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 
---
0x05 0x00 0x03 0x01 0x10 0x00 0x00 0xa3 0x05 0x00 0x21 0x00 0x41 0x0d 0x10 0x00 
0x05 0x03 0x01 0x00 0x00 0x12 0x10 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 
---
0x05 0x00 0x02 0x01 0x10 0x00 0x00 0xa3 0x0f 0x00 0x00 0x00 0x00 0x00 0x00 0x00 
0x05 0x00 0x07 0x00 0x00 0x12 0x10 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 
---
0x01 0x00 0x00 0x80 0x72 0xb0 0xe6 0xfe 0x7f 0x00 0x00 0x14 0xe1 0x0f 0x09 0x01 
0x01 0x00 0x07 0x00 0x00 0x12 0x10 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 
---
0x03 0x80 0x72 0xb0 0xe6 0xfe 0x7f 0x00 0x00 0x19 0xe1 0x0f 0x09 0x01 0x00 0x00 
0x03 0x00 0x07 0x00 0x00 0x12 0x10 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 
Error: invalid response during transfer (count = 0/2, status = 7)
```
I got this interesting error.. so, what mistake I have did.. now? I love this debugging!!...

ataradov, 2021-06-01:
Now you don't have a target connected to the debugger. You can remove the hex print. The debugger now works.

ataradov, 2021-06-01:
Wait, the target got detected. So something else is wrong.

Sreedharbot, 2021-06-01:
I have removed the -e command...

```
sreedhar@Sreedhars-MacBook-Pro edbg % ./edbg -b -p -v  -t samd11 -f sam_ba_Generic_D11C14A_SAMD11C14A.bin

---
0x00 0x01 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 
0x00 0x00 0x07 0x2e 0x31 0x30 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 
---
0x00 0x02 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 
0x00 0x00 0x07 0x2e 0x31 0x30 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 
---
0x00 0x03 0x2e 0x31 0x30 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 
0x00 0x00 0x07 0x2e 0x31 0x30 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 
---
0x00 0x04 0x2e 0x31 0x30 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 
0x00 0x05 0x31 0x2e 0x31 0x30 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 
---
0x00 0xf0 0x2e 0x31 0x30 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 
0x00 0x01 0x11 0x2e 0x31 0x30 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 
Debugger: ARM DAPLink CMSIS-DAP 1093000031389eb81139333437353533a5a5a5a597969906 1.10 (S)
Clock frequency: 16.0 MHz
---
0x03 0x20 0xda 0x85 0xe9 0xfe 0x7f 0x00 0x00 0x9f 0x8a 0x3a 0x06 0x01 0x00 0x00 
0x03 0x00 0x11 0x2e 0x31 0x30 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 
---
0x02 0x01 0xf0 0xd6 0x85 0xe9 0xfe 0x7f 0x00 0x00 0xa0 0xa3 0x3b 0x06 0x01 0x00 
0x02 0x01 0x11 0x2e 0x31 0x30 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 
---
0x04 0x00 0x00 0x80 0x80 0x00 0x20 0xda 0x85 0xe9 0xfe 0x7f 0x00 0x00 0xba 0x8a 
0x04 0x00 0x11 0x2e 0x31 0x30 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 
---
0x13 0x00 0x20 0xda 0x85 0xe9 0xfe 0x7f 0x00 0x00 0xc1 0x8a 0x3a 0x06 0x01 0x00 
0x13 0x00 0x11 0x2e 0x31 0x30 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 
---
0x11 0x00 0x24 0xf4 0x00 0x20 0xda 0x85 0xe9 0xfe 0x7f 0x00 0x00 0xcc 0x8a 0x3a 
0x11 0x00 0x11 0x2e 0x31 0x30 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 
---
0x01 0x00 0x01 0x20 0xda 0x85 0xe9 0xfe 0x7f 0x00 0x00 0xd8 0x8a 0x3a 0x06 0x01 
0x01 0x00 0x11 0x2e 0x31 0x30 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 
---
0x10 0x00 0x83 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0xa0 
0x10 0x00 0x11 0x2e 0x31 0x30 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 
---
0x10 0x80 0x83 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0xa0 
0x10 0x80 0x11 0x2e 0x31 0x30 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 
---
0x12 0x88 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0x9e 0xe7 0xff 0xff 0xff 0xff 0xff 
0x12 0x00 0x11 0x2e 0x31 0x30 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 
---
0x05 0x00 0x01 0x02 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 
0x05 0x01 0x01 0x77 0x14 0xc1 0x0b 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 
---
0x05 0x00 0x03 0x00 0x16 0x00 0x00 0x00 0x08 0x00 0x00 0x00 0x00 0x04 0x00 0x0f 
0x05 0x03 0x01 0x77 0x14 0xc1 0x0b 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 
---
0x05 0x00 0x03 0x01 0x12 0x00 0x00 0xa3 0x05 0x18 0x21 0x00 0x41 0x0f 0x00 0x00 
0x05 0x03 0x01 0x06 0x01 0x03 0x10 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 
Target: SAM D11C14A (Rev B)
---
0x05 0x00 0x03 0x01 0x10 0x00 0x00 0xa3 0x05 0x02 0x21 0x00 0x41 0x0f 0x00 0x00 
0x05 0x03 0x01 0x00 0x00 0x12 0x10 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 
---
0x05 0x00 0x03 0x01 0x12 0x00 0x00 0xa3 0x05 0xf0 0xed 0x00 0xe0 0x0d 0x03 0x00 
0x05 0x03 0x01 0x00 0x00 0x12 0x10 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 
---
0x05 0x00 0x03 0x01 0x12 0x00 0x00 0xa3 0x05 0xfc 0xed 0x00 0xe0 0x0d 0x01 0x00 
0x05 0x03 0x01 0x00 0x00 0x12 0x10 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 
---
0x05 0x00 0x03 0x01 0x12 0x00 0x00 0xa3 0x05 0x0c 0xed 0x00 0xe0 0x0d 0x04 0x00 
0x05 0x03 0x01 0x00 0x00 0x12 0x10 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 
---
0x05 0x00 0x03 0x01 0x10 0x00 0x00 0xa3 0x05 0x01 0x21 0x00 0x41 0x0d 0x00 0x02 
0x05 0x03 0x01 0x00 0x00 0x12 0x10 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 
Programming...---
0x05 0x00 0x03 0x01 0x12 0x00 0x00 0xa3 0x05 0x04 0x40 0x00 0x41 0x0d 0x00 0x00 
0x05 0x03 0x01 0x00 0x00 0x12 0x10 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 
---
0x05 0x00 0x03 0x01 0x12 0x00 0x00 0xa3 0x05 0x1c 0x40 0x00 0x41 0x0d 0x00 0x00 
0x05 0x03 0x01 0x00 0x00 0x12 0x10 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 
---
0x05 0x00 0x03 0x01 0x11 0x00 0x00 0xa3 0x05 0x00 0x40 0x00 0x41 0x0d 0x41 0xa5 
0x05 0x03 0x01 0x00 0x00 0x12 0x10 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 
---
0x05 0x00 0x03 0x01 0x10 0x00 0x00 0xa3 0x05 0x14 0x40 0x00 0x41 0x0f 0x00 0x00 
0x05 0x03 0x01 0x01 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 
---
0x05 0x00 0x03 0x01 0x11 0x00 0x00 0xa3 0x05 0x00 0x40 0x00 0x41 0x0d 0x02 0xa5 
0x05 0x03 0x01 0x01 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 
---
0x05 0x00 0x03 0x01 0x10 0x00 0x00 0xa3 0x05 0x14 0x40 0x00 0x41 0x0f 0x00 0x00 
0x05 0x00 0x07 0x01 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 
---
0x01 0x00 0x00 0x50 0xd2 0x85 0xe9 0xfe 0x7f 0x00 0x00 0x14 0x81 0x3a 0x06 0x01 
0x01 0x00 0x07 0x01 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 
---
0x03 0x50 0xd2 0x85 0xe9 0xfe 0x7f 0x00 0x00 0x19 0x81 0x3a 0x06 0x01 0x00 0x00 
0x03 0x00 0x07 0x01 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 
Error: invalid response during transfer (count = 0/3, status = 7)
```

Now I am facing.. this error 

## Error: invalid response during transfer (count = 0/3, status = 7)

ataradov, 2021-06-01:
Please don't add any more of thew hex stuff. It is no longer relevant.

Something is still wrong with the target device. Do you have a reset asserted by any chance?

Sreedharbot, 2021-06-01:
I didn't had any 
> reset asserted by any chance

moreover.. I tried to run. program  many times. and I got the result  "Error: invalid  response received" only..
and I didn't have any indication of reset assert..

ataradov, 2021-06-01:
I meant asserted by some external circuitry. Based on the previous logs, your SWD communication is working, edbg managed to read the device id and recognize the part.

But then programming fails. So double check your power supply to the target and electrical stuff like this.

And yes, a new chip test would be useful too.

Sreedharbot, 2021-06-01:
Hey..good news it is working.. & thanks  

Initially, I have powered from 5V usb then I have removed that connection and powered from particle debugger. and the output was success.. 

## Hey, if you are interested to share, how you were able to debug the issue. Please share.. ?? I'm very interested to known the backend process.. of this..??

What is USB FS device? , why i needed to do the change the value from 512 to 64.. and How did you conclude that there is some electrical issue.. (did you read from HEX code or ???) . 

## Please answer to the above question it will be helpful...

ataradov, 2021-06-01:
Don't do bold text either. I can read fine.

I was able to tell because I ran into a similar issue some time ago with another person. Hex codes were useful in confirming the issue. You could see that debugger responded with 0xff for all but first command. 0xff is the error code. It means that it was interpreting extra byes (stuff over 64 bytes) as commands too. And that obviously would not work.

USB HID protocol that is used for CMSIS-DAP communication requires frames to be sent with fixed size. USB Full Speed devices support 64 bytes max, so this is the size that is commonly used. USB High Speed devices can go higher, so 512 or 1024 bytes are common.

Linux and Windows ports use OS-specific APIs to determine the size of the endpoint and automatically adjust. MAC port is not really maintained, since I don't use macs and don't have any apple hardware. So the size is just set to a fixed value.

closed 2021-06-01 by Sreedharbot