ataradov/edbg #115: Ubuntu 20 edbg flash broken?

biboc, 2021-09-16:
```
$ ./edbg --help
CMSIS-DAP SWD programmer. Built Sep 16 2021 12:26:32.

$ git rev-parse HEAD
700b682a0c6c0df555bfa70aba9bc99f79dda07e

$ git status
Sur la branche master
Votre branche est à jour avec 'origin/master'.

$ ./edbg -t list
Supported device families:
  Atmel SAM C/D/L/R series:
    samd09 samd10 samd11 samd20 samd21 samc21 saml21 saml22 samr21 samr30 samr34 samr35 samda1 pic32cm
...

$ ./edbg -b -t samr30
Erreur de segmentation (core dumped)

$ uname -a
5.11.0-34-generic #36~20.04.1-Ubuntu SMP Fri Aug 27 08:06:32 UTC 2021 x86_64 x86_64 x86_64 GNU/Linux

$ lsb_release -a
No LSB modules are available.
Distributor ID:	Ubuntu
Description:	Ubuntu 20.04.3 LTS
Release:	20.04
Codename:	focal

```

So I can't access to my device

ataradov, 2021-09-16:
I just fixed one possible source of the segfalt. It actually happens when there is a problem accessing the debugger (permission error most likely)

I just happened to be on a fresh install and the issue reproduced easily.

See if this solves your problem.

biboc, 2021-09-16:
I added hidraw rules in udev and it worked
I don't have time to remove and test but I'm sure you found the problem :) 
Thanks

For those who asking which rules:
```
$ cat /etc/udev/rules.d/99-hidraw-permissions.rules 
KERNEL=="hidraw*", SUBSYSTEM=="hidraw", MODE="0664", GROUP="plugdev"
```
And reboot your computer (or logout/login)

ataradov, 2021-09-16:
Yes, I'm pretty sure it is the same issue.

closed 2021-09-16 by ataradov