ataradov/edbg #116: unable to unlock

JefkeB, 2021-09-20:
Hi,

When locking an ATSAMD21 with the edbg tool, removing power from debugger & processor, wait a moment, re-apply power i'm unable to unlock the device. When not removing the power in between the unlocking seems to work.

The used debugger is a st-link2 with CMSIS-DAP firmware on it. Al the other operations seem to be working.

```
locking
D:\>edbg -b -t samd21 -c 1000 -k
Debugger: X893 ARM CMSIS-DAP HID 0001A0000000 1.0 (S)
Clock frequency: 1.0 MHz
Target: SAM D21G18A (Rev D)
Locking... done.
```

```
unlocking
D:\>edbg -b -t samd21 -u
Debugger: X893 ARM CMSIS-DAP HID 0001A0000000 1.0 (S)
Clock frequency: 16.0 MHz
Error: invalid response during transfer (count = 0/1, status = 7)
```

Tried different Clock frequencies.

With j-link commander and a jlink mini the unlocking can be done.

What is the cause of this, and how to solve it ?

JvB

ataradov, 2021-09-20:
Unlocking on this device is the same as erase.

Do you have reset pin connected to the debugger?

JefkeB, 2021-09-20:
No reset pin is currently in use ...
that could be an issue ?

ataradov, 2021-09-20:
It might be. edbg uses reset extension mode to prevent code execution on the device after the reset, and that mode needs to have reset pin connected.

JefkeB, 2021-09-20:
Ok,

Added the 'reset' pin

And now it is possible to unlock the device !
D21 as well as a D51

Thanks for the very quick response

And thanks for this software !

JvB

closed 2021-09-20 by JefkeB