ataradov/edbg #117: Binary downloads

ooxi, 2021-11-19:
Since bintray has shut down, we no longer provide binary downloads. This is especially troublesome for Windows users.

With [gerbv](https://github.com/gerbv/gerbv) I have solved this problem by using GitHub actions to deploy to the build artifacts to a [separate repository](https://github.com/gerbv/gerbv.github.io) served by [GitHub Pages](https://gerbv.github.io/). It is not necessary to use a separate repository though, it could also be done by deploying the binaries to a `gh-pages` branch on the same repository.

Would you be interested in such a feature? I could provide a PR.

ataradov, 2021-11-19:
The binaries are not really missing, they are available as artifacts, so I just need to rebuild them every 3 months.

I generally like the idea of publishing them some place else. I need to look closer at how this works. I'd really like to find another simple service like bintray, but it does not look like anything like this exists, even in a self-hosted version.

ooxi, 2021-11-19:
> The binaries are not really missing, they are available as artifacts, so I just need to rebuild them every 3 months.

Yes they are not missing, they are just hard to discover.

> I generally like the idea of publishing them some place else. I need to look closer at how this works. I'd really like to find another simple service like bintray, but it does not look like anything like this exists, even in a self-hosted version.

For your information, I have created an example on how to publish using GitHub Actions to GitHub Pages (`gh-pages`) in [github-actions-release-on-github-pages-example](https://github.com/ooxi/github-actions-release-on-github-pages-example/blob/main/.github/workflows/ci.yaml).

The relevant part is this stage:

```
    - name: Deploy 🚀
      uses: JamesIves/github-pages-deploy-action@4.1.4
      with:
        branch: gh-pages # The branch the action should deploy to.
        folder: build # The folder the action should deploy.
```

This will publish the contents of the `build` directory (on GitHub Actions, thus this might contain build artifacts as well as a website template) to the `gh-pages` branch.

ataradov, 2021-11-19:
I had a long look at all this. And I don't think I want to support that stupid system where you run random people's code automatically and pretend like it is a good idea.

What is there now is fine until there are better CI and CD platforms.

This whole thing is super gross and I really don't want to be anywhere close to it.

ataradov, 2021-11-20:
Ok, I made my own bintray-like server. Actual file upload is just 100 lines of Python. I'll add index file generation, so it does not look like standard Apache index and run than on my server.

ataradov, 2021-11-20:
I've got it all mostly working. The static pages and files will be located here https://taradov.com/bin/edbg/ All the files there are junk right now. Once I fully debug the system, I'll nuke them all.

Automatic uploads for all 3 OSes were tested on this test repo https://github.com/ataradov/x-actions-test Build file there shows how the API looks like.

The script itself is here https://github.com/ataradov/badtray

Right now it is somewhat flaky and gets killed or dies unpredictably. I'll need to figure out why.

ooxi, 2021-11-20:
This is great new!

ataradov, 2021-11-20:
It looks like it runs fine now. I've added uploads to the new server. All images at https://taradov.com/bin/edbg/ are now valid builds.

closed 2021-11-20 by ataradov