ataradov/edbg #121: CMSIS-DAP v2 support requirement

llinjupt, 2022-02-21:
Hi ataradov,

Thanks for your greate work first. Would you like to support CMSIS-DAP v2 which is faster and becoming more popular? 
Anyway thx again.

ataradov, 2022-02-21:
There is a long term plan to add v2 support. But that would mean migration to libusb (and likely abandonment of the MAC OS support unless someone will want to work on it). I have some test software that does this already (libusb, but just for the HID part).

There are a couple of reasons why this is a lower priority thing. First of all, all my hardware is based on free-dap, so I would need to implement v2 there to benefit from this change. And then programming is mostly limited by  flash controller write speed, not by the transfer speed. Reading and verification would benefit more, of course.

And more importantly, I'm working on some other stuff that would need v2 support to, so it is likely that it would be implemented there first and back-ported here.

ataradov, 2022-02-27:
I was looking into that, but I'm finding it hard to get precompiled firmware for the debuggers I have that would support CMSIS-DAP v2. And specifically devices that support both.

What specific programmers you had in mind? If you have some, can you capture their complete USB descriptors?

llinjupt, 2022-04-07:
I have been trying to implement a V1/V2 compatilbe probe and it's almost done. BTW, the most important point which I have to mention is  it 's as fast as STLINKv3. But I am afraid that's implemented on C51 instead of STM32, so the firmware won't fit your hardware. Moreover, the english document is not finished yet.  As soon as it 's finished I will let you know which may help you somehow.

[image]

ataradov, 2022-04-07:
I have added v2 support into free-dap, so the only thing I'm missing now is a pre-built firmware for some standard board that I can use for cross-cheking.

ataradov, 2024-01-16:
CMSIS-DAP v2 support was added on Linux and Windows. MAC OS will have to deal with the slow stuff.

closed 2024-01-16 by ataradov