ataradov/edbg #124: edbg compiled on Raspberry Pi 4B hangs while using free-DAP to program SAMD21

mkindschi, 2022-05-05:
I am using edbg (compiled on a Raspberry Pi 4B 4GB) to program a SAMD21 MCU.
The Pi is connected to a Raspberry Pi Pico board running the latest Free-DAP.  When the debugger is called from the command line, it proceeds through any reset or erase sequence, and then starts, but does not complete programming.  It appears to just stop.  I have attached a Saleae logic analyzer capture (and a .csv export) of both a failed (Raspberry Pi -> pico) and successful (Windows PC -> pico) programming capture. CH0 is SWCLK, CH1 is SWDIO, and CH3 is nRESET.
There are no red flags in the dmesg log, and power integrity is not an issue either.  We can repeat this on multiple Pi's.
I would greatly appreciate any help, or known issues using edbg on Pi.
Let me know also if there is more specific debugging info I should gather.

Below is a sample of what I see when I attempt to program with edbg on the Pi.
>pi@raspberrypi:\~$ sudo edbg -x 100 -s A03FD687 -t samd21 -o 0 -z 261120 -bpv -f ./Downloads/TowerImage-6.08.38-debug.bin
Debugger: Alex Taradov Generic CMSIS-DAP Adapter A03FD687 2.0.0 (SJ)
Clock frequency: 16.0 MHz
Resetting... done.
Target: SAM D21J18A (Rev D)
Programming....**^C**
pi@raspberrypi:\~$

[EDBG_Fails_RPi_pico.zip](https://github.com/ataradov/edbg/files/8631711/EDBG_Fails_RPi_pico.zip)
[EDBG_success_PC_pico_short.zip](https://github.com/ataradov/edbg/files/8631896/EDBG_success_PC_pico_short.zip)

mkindschi, 2022-05-05:
The -c 100 option works successfully. Thank you.
Is it possible that it may be a pin drive issue, and that the Pico/RP2040 may need a buffer/driver to make the clock run at a higher speed (or achieve a higher slew rate)?  Looks like the GPIOs can be set to 2, 4, 8, or 12 mA, with a default of 4.

ataradov, 2022-05-05:
No, it is just a limitation of how clock is implemented in free-dap. There are two sets of clock functions - up to a certain frequency (~8.5 MHz) a version with the delay loop is used and the frequency adjustment is smooth. At 8.5 MHz we reach the limit of the loop (< 1 iteration needed). So I substitute a version without loop at all, but that causes a sudden jump in clock frequency, which target can't handle.

So just run it with "-c 8500" and you will get maximum "good" clock.

closed 2022-11-03 by ataradov