ataradov/edbg #127: board unusable after write fuse operation on SAML11-XPRO

dylad, 2022-07-27:
When trying to write fuse on SAML11-XPRO, operation looks successful but afterward the board is unusable. No more EDBG operations are possible. The only solution I found right now is to issue an ERASEALL cmd from Microchip Studio.
Read operation works flawlessly.

fuses read/write work fine on SAML10-XPRO.

Example:
```

 ~/work/tools/edbg  master  ./edbg -t saml11  -F r0,194 --verbose                                               INT ✘  22:37:42 
Debugger: ATMEL EDBG CMSIS-DAP ATML3138051800000299 03.25.01B6 (S)
Clock frequency: 16.0 MHz
Target: SAM L11E16A (Rev B)
Fuses:
  reading bit 194 from section 0: 0x0 (0)
done.
 ~/work/tools/edbg  master  ./edbg -t saml11  -F w0,194,1 --verbose                                                 ✔  22:37:51 
Debugger: ATMEL EDBG CMSIS-DAP ATML3138051800000299 03.25.01B6 (S)
Clock frequency: 16.0 MHz
Target: SAM L11E16A (Rev B)
Fuses:
  writing value 0x1 to bit 194 in section 0: OK
done.
 ~/work/tools/edbg  master  ./edbg -t saml11  -F r0,194 --verbose                                                   ✔  22:38:06 
Debugger: ATMEL EDBG CMSIS-DAP ATML3138051800000299 03.25.01B6 (S)
Clock frequency: 16.0 MHz
Target: SAM L11E16A (Rev B)
Fuses:
Error: invalid BootROM response 0x25, expected 0x39. Check that device is not locked.
```

ataradov, 2022-07-27:
This happens because user row CRC no longer matches. 

The erase command in this case also returns "STATUS = 0xec000011", which is "UROW checksum error".

This is probably the error I can ignore, in the chip erase erase so that at least recovery is possible.

I'll need to think what to do with the CRC part. I could recalculate it, but it would go against the policy of not modifying the supplied binaries. This policy does not make much sense in case of single bit modifications.

ataradov, 2022-07-27:
Well, it looks like for BOCOR I'd need to implement SHA-256 as well.  I don't want to get dependencies on external libraries, so I'll need to figure out how hard it is to code by hand. On a quick read of the FIPS 180-4, it seems pretty straightforward to do an non-optimized version. 

And yes, ignoring the error on erase works and recovers the device. I'll push that along with the CRC calculation.

ataradov, 2022-07-28:
Should be fixed now. By default fuse write operations would update CRC and SHA-256 hash. I added aliases to the same fuse sections that would write the data as is.

closed 2022-07-28 by ataradov

dylad, 2022-07-28:
Thanks for the super quick fix @ataradov !