ataradov/edbg #131: libhidapi.0.dylib not loaded on Mac

qbolsee, 2022-11-10:
Hi,

Trying to run  `edbg` on recent Mac systems produces the following error:

```
dyld: Library not loaded: /usr/local/opt/hidapi/lib/libhidapi.0.dylib
  Referenced from: ./edbg
  Reason: image not found
```

This seems to be specific to Apple M1 based machines. I'm guessing a manual installation of hidapi could fix it but I'm integrating `edbg` as part of this [Arduino core](https://github.com/qbolsee/ArduinoCore-fab-sam) to distribute it to a larger audience, so an all-in-one binary would be ideal.

ataradov, 2022-11-10:
I have no idea about that. I don't own a Mac or any Apple hardware. This was contributed by others. If you can figure out how to fix this - let me know or submit a pull request.

Ideally dbg_mac.c should not depend on hidapi and use whatever low level OS APIs are available. Or at least figure out a way to use a static library and include it with a build.

I'm not even sure why it worked before. We brew install the library for the build on the build machine, and I guess user machines all had it installed so far.

In any case, it is a dependency, and should be installed the same way as the build machine "brew install hidapi". But there is a real chance that it is not available for M1 or something like this.

qbolsee, 2022-11-10:
Thanks Alex, that's good insight. I'll try to get a Mac user to investigate; a static library would be ideal indeed.

ataradov, 2022-11-10:
Ideally we need someone with a bit of mac USB programming knowledge to update the whole driver. It is quite outdated in functionality. It is missing automatic detection of the end point size, so it works with a fixed set of debuggers and needs to be recompiled to support others. By default it uses report_size = 512, so it would not  work with simple SAM D11-based debuggers that use 64 byte reports.

wenhao-gao, 2022-11-11:
Hi, I also met the problem and found out the problem is caused by `pkg-config`. I ran `brew install pkg-config` then it's fixed.

ataradov, 2022-11-13:
Dependency on hidapi was removed, native APIs are used instead. Thanks to @qbolsee for providing access to a MacOS machine.

closed 2022-11-13 by ataradov