ataradov/edbg #134: "nanoDAP" adapter not supported in Linux

snematbakhsh, 2023-06-16:
I recently purchased several low-cost CMSIS-DAP adapters from China which seem to be based on the "nanoDAP" project (https://github.com/wuxx/nanoDAP). These adapters are detected fine in the Windows version of edbg (tested latest release binary) but not the Linux version.

Bus 003 Device 030: ID c251:f001 Keil Software, Inc. CMSIS_DAP
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               2.00
  bDeviceClass          239 Miscellaneous Device
  bDeviceSubClass         2 
  bDeviceProtocol         1 Interface Association
  bMaxPacketSize0        64
  idVendor           0xc251 Keil Software, Inc.
  idProduct          0xf001 
  bcdDevice            1.00
  iManufacturer           1 CMSIS-DAP by ARM
  iProduct                2 CMSIS_DAP

The issue is that the product string descriptor is (non-standard?) "CMSIS_DAP", not "CMSIS-DAP".

The following patch fixes detection:

diff --git a/dbg_lin.c b/dbg_lin.c
index 6045e5c..01e240b 100644
--- a/dbg_lin.c
+++ b/dbg_lin.c
@@ -68,8 +68,10 @@ int dbg_enumerate(debugger_t *debuggers, int size)
       debuggers[rsize].vid = strtol(udev_device_get_sysattr_value(parent, "idVendor"), NULL, 16);
       debuggers[rsize].pid = strtol(udev_device_get_sysattr_value(parent, "idProduct"), NULL, 16);
 
-      if (strstr(debuggers[rsize].product, "CMSIS-DAP"))
+      if (strstr(debuggers[rsize].product, "CMSIS-DAP") ||
+          strstr(debuggers[rsize].product, "CMSIS_DAP")) {
         rsize++;
+      }
     }

Shall I put up a PR?

The one mystery is how detection is working in Windows because the product comparison code in dbg_win.c looks very similar.

ataradov, 2023-06-17:
I'd rather not have workarounds for non-compliant hardware in the main code. 
I checked a few binaries in that repo, and I can't find anything that would have _ in the string. So, may be firmware update would bring them up to the standard.

snematbakhsh, 2023-06-18:
Hi, thanks for the reply, that sounds reasonable. I will close this issue.

closed 2023-06-18 by snematbakhsh