ataradov/edbg #135: WAIT response after erasing flash

urban-achiever, 2024-01-17:
Hi,

I'm using a custom board running free-dap to try to program a SAMRH71. After making some changes in target_atmel_cm7, edbg can see my device, and even read the CHIPID and all flash information. When it gets to programming my target, after erasing the first erase block, when reading FSR to see FRDY setting, the dap transfer returns satus 2 WAIT. Any idea what might cause this? I've set the flash address and size as per the datasheet. I've also checked the lock bits, they are all unset.

ataradov, 2024-01-17:
I have never worked with RH71, so I don't really know. WAIT means that the bus was busy. It is possible that this is how this device is designed.

Technically WAIT should not be the reason for the edbg to fail, it must retry the transfer. But this is rarely happens, so it is not implemented.

One thing you can try is add sleep_ms(10); after the command before reading the status bit. This is a hack and would slow things down, but it would make sure that device is ready by the time you start reading the status.

I'm doing something similar for the M480 target. But I don't remember what exactly was going on there. Quite possible the same thing.

urban-achiever, 2024-01-18:
The doc for the RH71 says that the device implements the standard Arm CoreSight macrocell, so there shouldn't be a difference I think?

I tried the delay, unfortunately it didn't help. I tried erasing only a single page, as well as erasing the enitre flash, but in both cases I get the same result. I've also realised that the bus is not immediately busy after the erase. It sometimes spends a few seconds in the while loop reading the FSR, but FRDY never sets, until eventually I get the WAIT ACK. The read command seems to work ok though.

I also tried attaching a jLink Pro, it could do write/erase/read without an issue.

ataradov, 2024-01-18:
It has standard debug components that do work as expected, since you can do anything at all. But CoreSight ends with the bus interface. What this bus does is specific to the deice. In this case bus seems to be locked, so CoreSight can't get access to it and returns WAIT.

I can't really help without access to the device and experimentation. 

One thing that may help is getting logs from the J-Link to see what it is doing. It may be something specifically fro this device. If you can, initiate the erase command and capture the SWDIO/SWCLK pins with a logic analyzer.

urban-achiever, 2024-01-18:
Ok I will see what I can do, thanks for the help!

ataradov, 2024-01-18:
And again, it is likely that J-Link actually handles WAIT as expected and retries the request. This is technically the correct way of dealing with this. I never implemented it, since it was not necessary.

urban-achiever, 2024-01-22:
So I figured out what the issue was. Apparently the flash controller must be clocked by the PMC at 2 MHz for the erase/write commands to work. This is in the datasheet, I just missed it. I added some clock configuration to edbg, and now I have no problem programming my target.

closed 2024-01-22 by urban-achiever