ataradov/edbg #136: Can't list devices on Ubuntu 20.04

strawberryhacker, 2024-01-20:
Hi,

I am using a fresh install of Ubuntu 20.04 with `libudev-dev` installed. 

Running `edbg -l` does not detect any programmers. 

Trying with sudo result in `Error: ioctl(CONTROL): -1`. 

I replaced the following line with `if (res < 0) return 0;`: https://github.com/ataradov/edbg/blob/c983cef053f4e64f5c8311dba541b35fa9af0d47/dbg_lin.c#L46 

Now it worked running the command as sudo. 

I have no idea why it happens, or if it is a real issue.

ataradov, 2024-01-20:
I will try to replicate this. What debugger was used?

As far as sudo part goes - did the same debugger work on the old version (before CMSIS-DAP v2 support) without sudo?

ataradov, 2024-01-20:
Also, does it fail the same way with sudo if debugger is not plugged in?

I think this may be due to some other device failing to sent the string descriptor.

closed 2024-01-20 by ataradov

strawberryhacker, 2024-01-20:
Sorry the late response. Ok, that seems reasonable. Thank you :-)