ataradov/edbg #139: Add support for PUYA PY32F0xx

ShmulikKraus, 2024-04-28:
Hi Alex, could you please take a look at adding the PUYA PY32F0xx series of MCUs to the supported targets? They're very affordable and quite powerful (for the price), especially the PY32F002A which is a $0.11 MCU ($0.08 in large quantities) with a great VFM. Even has a (very preliminary) Arduino support ( https://arduino.py32.org ).
You can find links to some resources such as English datasheets and reference manuals, utilities, etc. in the 1st post of this thread:
https://www.eevblog.com/forum/microcontrollers/$0-11-py32f002a-m0-24mhz-320kb-actually-324kb-more-peripherals/

PUYA also released a package of OpenOCD with cfg files that support the PY32F chips, not sure if it helps or not. You can download it here:
https://gitee.com/puya-semiconductor/tools-and-software/blob/master/PY32_GCC/openocd-0.12.0.zip

ataradov, 2024-04-28:
I will look into supporting them. They are very similar to STM32 parts, so overall support should not be too hard. But there are a number of annoying differences between the devices. PY32F002B does not seem to have any documented flash read protection, for example. And 004 devices have option bytes located at a different address.

I only have PY32F002A, so this is the only device I can test, so I will be targeting that.

ataradov, 2024-04-28:
I have added PY32F002A support. It should also work for PY32F002B, but I can't find device IDs for any of the devices, so I just added the one I have for tests. If you have any other devices, I can work with you on figuring out those IDs and adding them to the program.

Other devices are similar, but different enough that adding them is not as easy as adding device ID line. But it still should be possible to do remotely.

ShmulikKraus, 2024-04-29:
Wow that was quick! Many thanks Alex! I only have PY32F002A right now, but I ordered some  PY32F002BW15S6TU and just yesterday the package was shipped. Should be here in about 3 weeks or so.
When it does arrive, will using OpenOCD to program it and thus giving me the Chip ID be enough or should I do anything else?

ShmulikKraus, 2024-04-30:
No problem, and thanks again.

ShmulikKraus, 2024-05-25:
My package finally arrived. As promised here is the code for the PY32F002BW15S6TU:
0x20220064
[image]

You can close this issue now, if you want to.

ataradov, 2024-05-25:
This code was already added, the most recent version should recognize this device. It was reported on EEVBlog forum.

ataradov, 2024-05-25:
Closing the issue.

closed 2024-05-25 by ataradov