ataradov/edbg #140: Artery AT32 family

ACleverDisguise, 2024-11-05:
I'd like to contribute support for the Artery family of Cortex-M processors.  I have access to AT32F403ACG, AT32F403ARG, and AT32F403AVG boards as well as an AT32F435ZM board.  Looking over the RMs for these, it looks like I could easily add support for the entire AT32F403A (sic) and AT32F435 families.

If I knew how.

A document that explains in broad terms what kinds of things need to be supplied to add support for new hardware would go a long way toward me getting this to you.  I took a look into target_st_stm32g0.c (the supported target I'm most familiar with) and it doesn't look that hard, but a document that outlines the framework of such support would be real nice.

As an added enticement I can probably add STM32L0 and STM32L4 support at a later date.  (I have a more immediate need for AT32 support, so it would be the first one I tackle.)

ataradov, 2024-11-05:
I don't have specific instructions. Generally, a good approach is to take any of the already supported targets and implement all the functions in a way that works for the new device. The top level functions you have to implement are enumerated in the target_ops_t structure.

I do not accept contributions for devices I don't have hardware for, since it would be impossible for me to maintain the code in the future.

ACleverDisguise, 2024-11-06:
Would you accept contributions that turn the targets into separate shared libraries so that others can add hardware support that they themselves maintain?

ataradov, 2024-11-06:
I don't think so. I have another project in the works where that would be possible, but that's  not going to be ready any time soon.

It is fine to have a pull request in case others find it helpful. If there is a lot of demand I'll figure out a way to get the hardware and add that support. But I really don't want to add dozens of random Chinese ST clones that nobody uses.

closed 2024-11-06 by ACleverDisguise