ataradov/edbg #142: Not working with MCU Link debug probe (incorrect DAP_INFO_CAPABILITIES size)

martin-vitek, 2025-07-25:
I'm using [MCU Link](https://www.nxp.com/design/design-center/software/development-software/mcuxpresso-software-and-tools-/mcu-link-debug-probe:MCU-LINK) debug probe, which implements CMSIS-DAP protocol. It has the latest FW (v3.160) and it's working with OpenOCD. With `edbg` I'm getting following error:
```
$ edbg -t samd10 -e -b
Error: incorrect DAP_INFO_CAPABILITIES size
```

Programmer is identified:
```
$ edbg -l
Attached debuggers:
  0: O3EDIY1K1DDIN - NXP Semiconductors MCU-LINK (r0FB) CMSIS-DAP V3.160 (2)

$ lsusb
Bus 003 Device 008: ID 1fc9:0143 NXP Semiconductors MCU-LINK (r0FB) CMSIS-DAP V3.160
```

Let me know, if I can provide some additional information.

ataradov, 2025-07-25:
I think I have this debugger, things worked with the previous firmware. I'll update it later and try with the latest.
It is likely that you can just comment out the line "check(size == 1, "incorrect DAP_INFO_CAPABILITIES size");" in edbg.c. There is not a real standard for CMSIS-DAP, so a lot of those checks are just assumptions based on how early versions behaved. They likely extended size of the capabilities field. It should not affect operation if we simply ignore the new fields.

ataradov, 2025-07-25:
Just as expected, they extended capabilities field size from 1 byte to 2 bytes. I extended the check for that.